cartographer-project/cartographer_ros #1112: Using Cartographer in place of ACML

CycleMark, 2018-12-01:
Hi,

I've looked @ https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#pure-localization.

I believe I am correct in saying that once the map has been finalised and saved down. I can then reload it and then should be able to have the robot start up and publish it's sensor data. Then Cartographer should be able to work out where the robot is in the in map without out providing a prior. It that assumption correct?

How do I achieve the latter part? That is reload the map and know that Cartographer has worked out where the robot is in the environment?

Thanks and Regards

Mark

gaschler, 2018-12-03:
Yes, Cartographer does localization.

> Then Cartographer should be able to work out where the robot is in the in map without out providing a prior.

The global localization is orders of magnitude slower than the one with a prior, but is supported. The prior is called the initial pose feature.
Actually, https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#pure-localization is an example for global localization, but the map is small so it snaps in fast.

> That is reload the map and know that Cartographer has worked out where the robot is in the environment?

This is possible but a little involved:
You can query `PoseGraphInterface::constraints()` and check if a scan match constraint from your current trajectory to that of the map exists.

CycleMark, 2018-12-06:
Hi,

Thanks for the reply.

What am I missing here  - I am trying to get the map to load, with the pose_graph_filename from a previously saved PB file and the robot running and publishing data. All that happens is a new map with built and nothing of the original one.

roslaunch cartographer_turtlebot turtlebot_urg_lidar_2d.launch configuration_basename:=turtlebot_urg_lidar_2d.lua pose_graph_filename:=${HOME}//cartographer_v1.0/cartographer_full_downstairs_20102018.bag.pbstream

https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html

Seems to suggest the pb file and the bag file or the robots sensor data (in place of the robots data)

Kind Regards

Mark

closed 2018-12-17 by CycleMark