cartographer-project/cartographer_ros #1114: Compiling error with ceres solver

p1n0cch10, 2018-12-04:
Hi all,

I am on Ubuntu 14.04, and ROS indigo. I install the cartographer according to the link [here](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html), except the ceres solver.

When I `wstool update -t src`, it shows 'fatel: unable to access https://***: Failed to connect to ceres-solver.googlesource.com'. Hence, I download the ceres-solver [here](https://github.com/ceres-solver/ceres-solver/releases/tag/1.13.0), and unzip it and put it in 'cartographer_ws/src/.'

However, when I `catkin_make_isolated --install --use-ninja`, it shows that 

> ***/cartographer_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10445:35: error: variable or field ‘it’ declared void
> for (typename C::const_iterator it = container.begin();
> ninja: build stopped: subcommand failed.

the whole log for compiling is attached. Thank you all for your help.
[log.txt](https://github.com/googlecartographer/cartographer_ros/files/2642717/log.txt)
[log.txt](https://github.com/googlecartographer/cartographer_ros/files/2642720/log.txt)

p1n0cch10, 2018-12-04:
For Eigen installation, I downloaded the source code [here](https://github.com/eigenteam/eigen-git-mirror), and:

> mkdir build
> cd build
> cmake ..
> make install

cesarhcq, 2019-06-21:
First, you need to install ceres-solver in your computer -> [here](http://ceres-solver.org/installation.html#linux).
Second, you need to update your Eigen to 3.3.7 -> [here](http://eigen.tuxfamily.org/index.php?title=Main_Page).

After this two steps, try to install cartographer again and it will work fine!

closed 2020-10-23 by MichaelGrupp