cartographer-project/cartographer_ros #1115: How to load a map in the cartographer for the resumption of the cartographer in uncharted areas

amburkoff, 2018-12-04:
On the first day, using cartographer, I create a corridor map and save it via map_saver.
On the second day, I need to load the map into the cartographer and continue creating the map in the unexplored rooms.
How to save a map from cartographer so that it can be loaded back into it and continue research in unknown areas?

rdesc, 2018-12-04:
**Step to save a map in a serialized state**
  1. ``rosservice call /write_state "filename: '$MAP_NAME.pbstream' include_unfinished_submaps: false"`` (**TIP:** press tab after ``/write_state`` to get arguments and note map gets saved by default to ``~/.ros``)

Map_saver only gives you the occupancy grid map (made by the cartographer_occupancy_grid_node) which is used purely for robot navigation stacks.

koki-ogura, 2018-12-10:
Its `load_state_filename:='$MAP_NAME.pbstream'`

Sanderi44, 2018-12-20:
One thing that could be useful would be to create a "LoadState" service that runs in the cartographer node.  This service would simply call the Node::LoadState method.  What I am unclear about is what will happen to the existing trajectory if LoadState is called in the middle of a trajectory.  My expectation is that it will overwrite that trajectory and start from scratch with the loaded state.  I will do some experimentation and report back.

Maibaf, 2020-02-29:
It seems, that is is not possible to resume mapping with an *.pbstream* file. **Is this correct?**

The result i get, in 2D SLAM on loading the state file, is, that the cartographer overrides the existing map. Mapping starts at the same point on the loaded map (i think this is the (0/0) point) like in the previous ride.
If the cartographer would perform a global localization first on the loaded map, continuing the mapping would may be possible. Does anyone know more about this?

nickvaras, 2020-05-05:
A trajectory should be finished before serializing to .pbstream. When mapping is restarted (and on the same starting point), a new trajectory is started.

Ga0aG, 2020-09-14:
> append load_state_filename to args.
> `<node name="cartographer_node" pkg="cartographer_ros" type="cartographer_node" args="-load_state_filename $MAP_NAME.pbstream" />`

I tried it, but cartographer_ros only load a frozen trajectory, there is no prebuilt map.

MichaelGrupp, 2020-10-23:
Original question was answered.

closed 2020-10-23 by MichaelGrupp