cartographer-project/cartographer_ros #1116: Scan matcher - out of tuning ideas

mojovski, 2018-12-05:
I would like to disable the global loop closing in the "pure localization" mode and only rely on the scan matcher.
This however, fails at "snipping" on the map.
[image]

This is my config file:
https://gist.github.com/mojovski/536493d912e2c698e20f5df66367a09f

Can somebody have a short look at the config file please? 

From my current understanding, the most important parameters should be
```
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window = 5.45
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.angular_search_window = math.rad(10.0)
```
But this does not work even if the offset between measured wall and the map-wall is smaller than 5.45m

mojovski, 2018-12-05:
Answering the last question: See here about correlative matcher as preprocessing for ceres scan matcher: https://github.com/googlecartographer/cartographer_ros/blob/80e4b0826b98dc77172f8fb0b8c0d26c3fcc2348/docs/source/tuning.rst#the-choice-of-scan-matchers

gaschler, 2018-12-05:
What do you mean by "snipping on the map"?

I took a look at the lua file.

> POSE_GRAPH.constraint_builder.min_score = 0.65

Slightly too high because this is above the histogram you show above.
Try 0.5--0.55 for your score histogram.

> POSE_GRAPH.constraint_builder.global_localization_min_score = 0.1

Far too low, would spend excessive computation time and find wrong matches.
You say you want to turn off global loop closing, so set this to 0.999

> POSE_GRAPH.overlapping_submaps_trimmer_2d

Don't use this for localization, use the localization trimmer instead.

> TRAJECTORY_BUILDER_2D.submaps.num_range_data = 2

Almost certainly too low. Need to be the number of incoming laser scan messages per 5--30 seconds.

> TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true

Is not a good choice in most cases. Correlative scan matching with its larger search area is only useful and necessary when a robot rotates far or moves extremely far between two laser scan rotations.

In general, your lua file has too many changes, it is usually better to tune one or two parameters at a time to stay close to the optimum.

Hope this helps.

ojura, 2018-12-05:
Set the sampling ratio in constraint builder to zero if you want to disable loop closure.

mojovski, 2018-12-05:
Hi @gaschler ,
thank you very much for your hints!

* Which modules is responsible for matching as a result for the constraints builder? Namely for generating these histograms? How can I tune this?

* You said, that using ```POSE_GRAPH.overlapping_submaps_trimmer_2d``` for localization is not a good idea. But what is the localization trimmer?

mojovski, 2018-12-05:
Is there a possibility, that if the computer is too slow, that the loop closing is stalling since all loop closing tasks overfill the stack of the global slam thread?

gaschler, 2018-12-05:
>  loop closing tasks overfill the stack of the global slam thread

Yes, loop closing can be delayed by long times because it is queued as background work.
You can register a `GlobalSlamCallback` and see how for the queue got processed.

closed 2020-10-21 by MichaelGrupp