cartographer-project/cartographer_ros #1118: Compiling error with abseil

carl1218, 2018-12-05:
Hi,
I am on Ubuntu 14.04, and ROS indigo,However, when I catkin_make_isolated --install --use-ninja, it shows this error:
`ninja: error: 'abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a' , needed by 'cartographer.common.blocking_queue_test' , missing and no known rule to make it`
What is the reason?
can anyone help me, thnak you  very much.

ojura, 2018-12-05:
Do you have ros-*-abseil-cpp installed? 

https://github.com/googlecartographer/cartographer_ros/issues/1075

carl1218, 2018-12-05:
@ojura 
thank you for your reply.
yes,i have installed it.so i don't konw what 's the reason.

carl1218, 2018-12-05:
@ojura 
thank you.
I have another question.
I installed the cartographer and cartographer_ros according by  the official process, but I found  the installed version was not the newest version (the code is different from the github), the previous version did not have abseil, SO, I git clone the cartographer, and then recompiled it,than it shows that error.

I want to konw ,why i build the cartographer is not a newest version,Is it for OS reasons?

ojura, 2018-12-05:
https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros.rosinstall

The rosinstall file specifies cartographer and cartographer_ros at 1.0. You can change that to master manually, or just go to the repos and checkout the current master manually.

closed 2018-12-06 by carl1218