cartographer-project/cartographer_ros #1122: Pure Localization Demo 2D

christqoh, 2018-12-07:
When trying to run the pure localization demo 2D from [here](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html) I get `Key 'pure_localization_trimmer' was used the wrong number of times.` (I first encountered this issue when trying to run pure localization on [my own code](https://github.com/Roboy/cartographer_ros/blob/roboy/cartographer_ros/configuration_files/roboy_localization.lua), where your updated syntax on pure localization didn't seem to work.)

Are you able to reproduce the behavior of the demo (let's not talk about my code for now) mentioned above? 

Best,
Chris

MichaelGrupp, 2018-12-11:
Which version of Cartographer are you using?

If you use the ROS Debian package, I think it's outdated and doesn't have this option yet.

If you run from source you should have this commit: https://github.com/googlecartographer/cartographer/commit/0981620d8fd4d3489ff1b503183b128909f1c86d#diff-d92a289a3dfb8172bab81b18f8adc359

MichaelGrupp, 2018-12-11:
Nevermind. I just didn't see that you linked your config.

Replace `TRAJECTORY_BUILDER_2D.pure_localization_trimmer = {` with `TRAJECTORY_BUILDER.pure_localization_trimmer = {`. The trimmer is not specific to the 2D pipeline.

christqoh, 2018-12-11:
Hi Michael,

I am using ROS Kinetic and installed cartographer_ros according to [your tutorial](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html) about three weeks ago so I should have that commit. 

Thanks for the hint with my implementation, but that was just a trial starting from [backpack_2d_localization](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_2d_localization.lua) as that [demo](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#pure-localization) is not working for me.

As mentioned above, do you have any idea why the [demo](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#pure-localization) isn't working?

MichaelGrupp, 2018-12-11:
The configuration of the demo looks fine (I didn't try it though). Are you sure it doesn't work with the upstream repository or could it be the changes you did in your fork? 

Because the backpack demo config was changed in your repository:
https://github.com/Roboy/cartographer_ros/blob/0697bcb9029ab9f7e47ea2b74a3ad8fc197bb963/cartographer_ros/configuration_files/backpack_2d_localization.lua#L17-L20

Compare that to [upstream](https://github.com/googlecartographer/cartographer_ros/blob/bb4aec395fea09609da64d4716b18072367933c8/cartographer_ros/configuration_files/backpack_2d_localization.lua). The `TRAJECTORY_BUILDER.pure_localization` parameter doesn't exist anymore, you have to use what you commented out.

MichaelGrupp, 2018-12-11:
One more question: how did you install the Cartographer library ([this repo](https://github.com/googlecartographer/cartographer)) and at which revision is it? Because it's very likely impossible that the old parameter works unless you linked with an outdated version of the library.

christqoh, 2018-12-11:
When installing `cartographer_ros` according to the [building and installation guide](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#building-installation) there is a line of code that reads:
```
wstool merge -t src https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/cartographer_ros.rosinstall
```
This calls a [cartographer_ros.rosinstall](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros.rosinstall) file which contains a line of code reading 
```
- git: {local-name: cartographer, uri: 'https://github.com/googlecartographer/cartographer.git', version: '1.0.0'}
```
So that's where I get the `cartographer` from I'd say with revision `1.0.0`.

MichaelGrupp, 2018-12-11:
@gaschler Can someone of you please update the documentation for the source installation? Apparently it doesn't set the `cartographer` version to master, as it is done in CI. It currently defaults to the 1.0.0 tag, which is not in sync with `cartographer_ros`' latest master.

@christqoh Quick fix: go to your `cartographer` clone and checkout the latest master, then re-compile.

christqoh, 2018-12-11:
Yep, that solved it. Thank you very much @MichaelGrupp!

christqoh, 2019-01-18:
Just a quick update: this does not work anymore. Checking out an older cartographer revision (such as [a33ef5d](https://github.com/googlecartographer/cartographer/tree/a33ef5de86306380663ecb476d187c71e35d8370) (which was right before this discussion) results in an error.
```
F0118 12:56:34.248379 19796 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'tsdf_range_data_inserter' was used the wrong number of times.
[FATAL] [1547812594.248580588]: F0118 12:56:34.000000 19796 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'tsdf_range_data_inserter' was used the wrong number of times.
```

hbx233, 2019-04-13:
Hi, I got the same problem and I checkout to the latest branch on remote (origin) and recompiled everything, but it still yield the same error, any suggestions? Thanks!

christqoh, 2019-04-13:
By referring to “latest branch” you mean cartographer? So I figured that the syntax used in the current demo file is different from the syntax used in the cartographer v1. I solved this by going into the history of the demo file and from there getting the old syntax. No manual checking out branches of cartographer or anything else, just compiling as is and getting the old syntax from the git history solved it for me.

msadowski, 2019-05-01:
I was getting the same error and using @christqoh pointer a checked the revision. Changing the lines referenced in this issue to `TRAJECTORY_BUILDER.pure_localization = true` indeed fixed it for me.

artemiialessandrini, 2019-10-25:
Is there any proper workaround of this issue?
It seems to be a serious one.

pmusau17, 2020-10-02:
Alright I found a workaround. Not gonna lie they could have just updated their website and it would have been all good. Anyway.... 
1. Do not execute this line: wstool merge -t src https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/cartographer_ros.rosinstall
2. instead create your own cartographer_ros.rosinstall file with the following: 
- git: {local-name: cartographer, uri: 'https://github.com/cartographer-project/cartographer.git', version: 'master'}
- git: {local-name: cartographer_ros, uri: 'https://github.com/cartographer-project/cartographer_ros.git', version: 'master'}
- git: {local-name: ceres-solver, uri: 'https://ceres-solver.googlesource.com/ceres-solver.git', version: '1.13.0'}
and then execute wstool merge -t src path/to/your/cartographer_ros.rosinstall
3. Change the cartographer_ros.rosinstall file within the folder src/cartographer_ros to reflect the above changes
4. Follow the rest of the [instructions](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html) and it should work
5. source/install_isolated/devel/setup.bash 

The demos should work now...smh

MichaelGrupp, 2020-10-22:
There is a actually a hint in the documentation: "If you build cartographer from master. Change/remove the version in the cartographer_ros.rosinstall." However, I think it makes much more sense to default to master. I will make a PR to change that.

> Just a quick update: this does not work anymore. Checking out an older cartographer revision (such as a33ef5d (which was right before this discussion) results in an error.

cartographer & cartographer_ros always need to be in sync, otherwise you can have API/ABI changes that are not compatible.

commit 1c9274ff (2020-10-22): Default to master in rosinstall (fixes #1122). (#1530)

This makes much more sense than 1.0.0, which is counter-intuitive.

If someone is building from source they usually do it to build the
latest bleeding-edge version.
closed 2020-10-22 by MichaelGrupp