cartographer-project/cartographer_ros #1125: 3D Map is Hazy; Transforms (Esp. odom) Is Unstable -- Jumps a Lot

kevinatorchen, 2018-12-11:
We have successfully used Cartographer for 2D SLAM on our Fetch Robot, but we are having difficulty with 3D SLAM. There are two problems we are encountering:

1. Map is Hazy/Unconfident. Below is a screenshot of the extremely hazy map generated:
[image]

2. The transforms are unstable -- they jump around a lot while we smoothly/slowly drive the robot. Odom jumps by far the most compared to the other frames.

To help debug this issue, I have uploaded the following files:

1. Rosbag Validate Output: https://gist.github.com/kevinatorchen/6c2c91ecafcd47fb0ba8e1c0dd5bd511
2. Bag file: https://drive.google.com/file/d/1YHZqJz7G_osJeDpRUtCRzLeCBMKXGBs-/view?usp=sharing
3. Pbstream file of map: https://drive.google.com/file/d/1uQ2dRuijEIT08MplY3iU-6uROUnGvOib/view?usp=sharing

4. Lua File (any fields omitted in this file use default values provided in trajectory_builder_3d.lua):

```include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 150
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.occupied_space_weight_0 = 1
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.occupied_space_weight_1 = 6
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.translation_weight = 15
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.rotation_weight = 5
TRAJECTORY_BUILDER_3D.submaps.high_resolution = .25
TRAJECTORY_BUILDER_3D.submaps.low_resolution = 1.0
TRAJECTORY_BUILDER_3D.submaps.num_range_data = 120

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 0 -- Disabled Global slam. Original value: 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options
```

Thank you for your help.

gaschler, 2018-12-12:
I took a very quick look at your validate output.

> E1211 15:38:50.815511 22742 rosbag_validate_main.cc:350] frame_id "laser_link" on topic base_scan_raw has serialization time 1544560356.353172561 but sensor time 1544560350.973716169 differing by -5.37946 s.

We should really add this explanation to the validate tool: While Cartographer only looks at header timestamps, cartographer_ros extrapolates to match your current clock (which is the serialization time in case of rosbag play with simulated clock), so its output is bogus (IMU or inertia extrapolated by 5 seconds).
When recording a bag, record it on the same OS where the sensors are connected.
Or, synchronize OS clocks.

> E1211 15:38:50.815928 22742 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "base_link". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)

Is this the expected? It is only fine if this is realtime output of Cartographer.
This bag file cannot be used for offline/simulation. As the error message says, this usually leads to fighting tf frames.

Another question, what kind of 3D sensor do you use? Cartographer takes expects little noise in depth measurements, so noisy depth images are not well suited.

kevinatorchen, 2019-01-17:
Thanks for your help, and sorry for the delay in response. Here are the answers to the questions you asked:

The error you addressed with "input contains transform message from frame_id 'map' to child_frame_id 'base_link'" is the realtime output of Cartographer. I accidentally included that in the bag file when creating this issue.

We use an RGB-D camera that is found on Fetch Robots: a "Primesense Carmine 1.09 short-range RGBD sensor"

MichaelGrupp, 2020-10-23:
>  noisy depth images are not well suited

> "Primesense Carmine 1.09 short-range RGBD sensor"

This seems to be the problem. Cartographer works best with LiDARs, if you use an RGB-D camera the outcome is usually not great.

closed 2020-10-23 by MichaelGrupp