cartographer-project/cartographer_ros #1128: 2D pure localization: no matched laser scan showing

NTUwilliam, 2018-12-12:
Hi all!

I've been using Cartographer for 2D SLAM for several months. Recently I'm trying to implement **2D pure localization**. 
I've successfully run the demo of 2D pure localization but when I'm trying to run my own rosbag, there's two issues:

**1.** No matched laser scan shown in the rviz and there's warning (no errors) related to **IMU**:
(1) The warning is:
[image]

(2) The rviz interface:
[image]

**2.** Why the map generated is not aligned with the grid line (as shown in the previous rviz interface)? Is it also something related to my **IMU** ?
What I want is something like this example (same X/Y direction with the grid):
[image]

Here's what `rosbag_validate` give me:
(1) For the rosbag used to **generate the map**:
[image]

(2) For the rosbag used to **run pure localization**:
[image]

Here's what `rosbag_info` give me:
(1) For the rosbag used to **generate the map**:
[image]

(2) For the rosbag used to **run pure localization**:
[image]

Here's my configuration files:
1. [offline_backpack_2d.launch](https://github.com/NTUwilliam/Google_cartographer/blob/master/offline_backpack_2d.launch)
2. [backpack_2d.urdf](https://github.com/NTUwilliam/Google_cartographer/blob/master/backpack_2d.urdf)
3. [backpack_2d_localization.lua](https://github.com/NTUwilliam/Google_cartographer/blob/master/backpack_2d_localization.lua)
4. [backpack_2d.lua](https://github.com/NTUwilliam/Google_cartographer/blob/master/backpack_2d.lua)
5. [demo_backpack_2d_localization.launch](https://github.com/NTUwilliam/Google_cartographer/blob/master/demo_backpack_2d_localization.launch)

Do you have comment or advice for these two issues? I can upload my rosbag files if needed.
Thanks in advance!

@gaschler @MichaelGrupp @ojura @cschuet

NTUwilliam, 2018-12-14:
Hi @Ricolus. Thanks for replying.

I removed the /tf topic from my rosbag (the rosbag used to generate the map / run pure localization / both of them). Then I got the totally misordered **.pbstream** like this:
[image]

Even the **.pbstream** is wrong so of course there's still no matched laser scan showing in the pure localization. And I still got repetitive warning saying that 
`[ WARN] [1544759056.892053210, 1532513080.641183901]: W1214 11:44:16.000000 31588 ordered_multi_queue.cc:155] Queue waiting for data: (1, imu)`

Any comment or advice?
Thanks!

MichaelGrupp, 2018-12-14:
> No matched laser scan shown in the rviz and there's warning (no errors) related to IMU

"Queue waiting for data: (1, imu)" means that no IMU data was received yet. In your [offline_backpack_2d.launch](https://github.com/NTUwilliam/Google_cartographer/blob/master/offline_backpack_2d.launch) you have specifed a topic remapping for the IMU, but you don't have it in your [demo_backpack_2d_localization.launch](https://github.com/NTUwilliam/Google_cartographer/blob/master/demo_backpack_2d_localization.launch).

> Why the map generated is not aligned with the grid line (as shown in the previous rviz interface)? Is it also something related to my IMU ? What I want is something like this example (same X/Y direction with the grid)

The grid in RViz is aligned with the TF frame that is configured as its "Reference Frame" in RViz (probably its the map frame). Apart from that, I don't quite understand what you expect. Do you want the walls of your map aligned with the grid? The map frame is not aligned with walls.

ojura, 2018-12-15:
The map is aligned with the heading of the beginning of the first trajectory. This cannot be changed.

If you want to have a fixed frame aligned to the walls, I suggest you calibrate the rotation manually and publish a fixed transform (rotation) from e.g. “map_aligned” to Cartographer’s “map” frame, and use “map_aligned” as the fixed frame in rviz.

NTUwilliam, 2018-12-17:
@MichaelGrupp 
Thanks for your advice. It works now! I did forget to specify a imu topic remapping in `demo_backpack_2d_localization.launch`. 
Actually I was planning to use the map built by Cartographer for laser scan pure localization in another algorithm called `mrpt_localization`. In that algorithm, the submitted map must have same X/Y direction with the grid. Since I successfully made Cartographer pure_localization work, this question no longer exists.

@ojura Thanks for your notes. Now I'm just using Cartographer pure_localization and no longer consider other algorithms so the fixed frame question doesn't exist anymore.

closed 2020-10-21 by MichaelGrupp