cartographer-project/cartographer_ros #1136: Getting odometry messages from Cartographer

cmoxham, 2018-12-20:
Hi there, 

Currently using cartographer_ros successfully (mostly) with an Ouster OS-1 lidar for mapping. I would like to integrate this into the ROS navigation package. This package requires odometry messages (both pose and twist). Is there a method to have cartographer publish this message type? Is this data even available from cartographer? A reply to #617 makes it sound like the /tf topic has this information.

Can anyone confirm that /tf has both pose and twist information? And is there an established method in ROS to take transform data and convert it into another message type? My understanding is that the /tf topic is not the same as other message types. 

Thanks,

sebastianklose, 2018-12-21:
Hi cmoxie,

we don't publish twist information from cartographer. The navigation stack is probably referring to wheel odometry (Look [here](http://wiki.ros.org/navigation/Tutorials/RobotSetup?action=AttachFile&do=view&target=overview_tf.png)).

So the question is, if your platform has odometry drivers.

The TF topic is essentially an array `geometry_msgs::TransformStamped` and the requirement from the navigation stack is, that it includes e.g. odom->base_link and additionally receives the nav_msgs/Odometry over the odom topic (which then includes the twist).

LMK if that helps.

cmoxie, 2018-12-21:
sebastianklose, Thanks for the reply! 

It does seem like the navigation stack is referring to wheel odometry, that seems to be what most people use. However, in our case, this is not possible. Wheel encoders would not be effective in our use case. It is my understanding that there are other methods to get odometry, like for instance, SLAM. 

Thanks for the clarification about the TF topic, that makes it a little more clear. So I guess to clarify my question, is it possible to get the nav_msgs/Odometry from cartographer? I'm pretty sure it doesn't publish those messages by default, but is the necessary information available? 

A more general question might be, shouldn't a SLAM package, like cartographer, output odometry information? My understanding is that it should be possible to perform SLAM, and navigation, using _only_ data from a lidar (pointcloud).

AlexReimann, 2018-12-27:
Maybe take a look at #1099, which adds a pose message.
SLAM gives you a pose estimate. If you really want to get also a twist, you could get some estimation by the change of pose.. not sure how good this will work though

hachbani2019, 2019-06-28:
Hello @cmoxie, i've been struggling to get mapping to work on my Ouster OS-1 using cartographer for like a month now, can you please share with me you .lua and launch files please ? It'd me help a lot to find the whereabouts of my problem. 

thanks