cartographer-project/cartographer_ros #1137: The frequency of Tf topic published is unstable and decreased

zldodo, 2018-12-23:
Hello,

I've working with Cartographer 2D for mapping and localization. When I ran the demo bag file(cartographer_paper_deutsches_museum.bag) for mapping, I found the frequency of tf published was unstable and decreasing/
[image]
[image]

This situation behaved worse with my bag( using Velodyne64 without IMU).
This situation is normal ?OR  it depends on hardware or something else? I have no idea that where the frequency of TF  is set.
In addition, for localization, if IMU was not available, would Cartographer work well? Do you have some test results about the localization accuracy in this case? 

Look forward for your help! Thank you!

gaschler, 2019-01-02:
No, by default, Cartographer ROS should publish at a fixed rate (even if it is waiting for input).

Are you sure you are measuring only Cartographer's own output and not including other tf messages, for instance those that may be inside a bag file you play?

zldodo, 2019-01-03:
I'm sure that there is no tf messages inside bags.  The case of decreasing rates was not found when using Offline_node.  As  you said, it is published at a fixed rate. 

Could you tell where the fixed rate is set？ in .lua file? pose_publish_period_sec = 5e-3

gaschler, 2019-01-03:
Correct, pose_publish_period_sec.

closed 2020-10-23 by MichaelGrupp