cartographer-project/cartographer_ros #1138: Fusion IMU magnetometer data

zhamengkeji, 2018-12-24:
Because it is necessary to extract the coordinates of the real world corresponding to the track points, I used a 9dof IMU mpu9250  and added a magnetometer data. I have two questions：1 Does the cartographer support the data of compass ? 2. Can the cartographer correct the map angle to the world map coordinates by the  data of compass? Thanks

gaschler, 2019-01-02:
> 1. Does the cartographer support the data of compass ?

Currently not.

> 2. Can the cartographer correct the map angle to the world map coordinates by the data of compass?

This depends what your goal is. You can always work around and publish a z-rotation to tf to rotate the map. (Cartographer does not assume that frames are static.)

Duke-Allen, 2019-03-11:
@zhamengkeji 
Hi, I am running cartographer with a IMU and a 16-lines lidar. I have some questions. 
My IMU is 6dor IMU MPU6050. Do I need to deal with the data from IMU? Is this IMU suited to 3D cartographer? 
I try to run the 3D cartographer with lidar and untreated data of IMU. But the result is so terrible. My robot model rotates all the time, do you know the problem?

closed 2020-10-23 by MichaelGrupp