cartographer-project/cartographer_ros #1140: WARN: dropped  earlier points

carl1218, 2018-12-25:
Hi,
 I used two 2d lidars (one 40hz,one 10hz) in cartographer 3d, I kept two lidars still and not moving.than i runned the launch ,i got this warn :`WARN: dropped  earlier points`,and  the map data is very confusing,and Laser data drift constantly.Is it the lidars problem?
[image]

this is rosbag：
https://drive.google.com/file/d/1G1h6-WhI09pvnxZTod9HgoBNuYK0w0Xf/view?usp=sharing
launch:
https://drive.google.com/file/d/1n6UG_JsHYpDKqr1Ft-0-Ie2DjJOaxdLG/view?usp=sharing
lua:
https://drive.google.com/file/d/1h-2d1RuM2iTCfG2070rtKOPbKYF0bA5Z/view?usp=sharing
urdf:
https://drive.google.com/file/d/1QJZH94aXWygD9bXrlwz65W1X0nQSyCXB/view?usp=sharing

gaschler, 2019-01-02:
Having two sensors with different rates is not fully supported, you may need to chunk and merge the messages yourself.

The best fix would be to tell the device drivers to send tiny messages and increase the accumulation number in cartographer.

> TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 2

You can try accumulating 5 messages to have 100 ms worth of data.

Also, please post the text output of the cartographer_rosbag_validate tool to rule out other things.

closed 2020-10-21 by MichaelGrupp

Gunz-katomaran, 2022-11-07:
haha.. you have to clear the issue by logical not commenting the code man @leonsuetfeld