cartographer-project/cartographer_ros #1143: Issue with Odom - Fatal failure when using odometry when running simulation in Gazebo

YizhouGaoVictor, 2018-12-28:
Hi I am recently trying to learn how to use Google Cartographer. While I am trying to get Cartographer to work with Gazebo, I ran into the following issue:
When setting use_odometry = false, I was able to register scan and create submap in rviz. However, without odometry, the quality of the map is horrible.

Thus when I try to include odometry in the Cartographer, it failed. The error code is attached.

When use_odometry = true;
[FATAL] [1546011133.029037595, 80.209000000]: F1228 10:32:13.000000 22824 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first (621355968800740000 vs. 621355968800740000) 
[image]

Also, the rqt_graph looks like this when not using odometry:

[image]

I was using cartographer_turtlebot branch. My package can be found at:
https://github.com/YizhouGaoVictor/turtlebot_slam.git

gaschler, 2019-01-02:
Cartographer expects incoming sensor data to have strictly increasing header timestamps (for each sensor independently, it does the merge sorting "collating" itself).
Your incoming odometry data seems to have duplicate timestamps.
(Perhaps the simulated odometry is faster than the simulated clock?)

YizhouGaoVictor, 2019-01-06:
Thank you Gaschler for replying my issue.

Inspired by your reply, I investigated the potential problem with timestamps of the odometry. What I found is that, the /odom topics I used to pass odometry data to cartographer happened to be the same topic that cartographer used to published odometry information for loop closure.

What I tried is to changed the tracking and publishing frame to be base_link to avoid publishing to /odom from cartographer.
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,

However, with this fix, even though the original problem is removed, I am left with the following warning:
[ WARN] [1546745794.174212670]: W0105 22:36:34.000000 18208 ordered_multi_queue.cc:155] Queue waiting for data: (0, scan)
[ WARN] [1546745794.218335014]: W0105 22:36:34.000000 18208 ordered_multi_queue.cc:155] Queue waiting for data: (0, scan)
[ WARN] [1546745794.308416581]: W0105 22:36:34.000000 18208 ordered_multi_queue.cc:155] Queue waiting for data: (0, scan)
[ WARN] [1546745794.350262395]: W0105 22:36:34.000000 18208 ordered_multi_queue.cc:155] Queue waiting for data: (0, scan)
[ WARN] [1546745794.394361959]: W0105 22:36:34.000000 18208 ordered_multi_queue.cc:155] Queue waiting for data: (0, scan)

It sounds like it never received the /scan topics. However, when I run rostopic echo /scan, I can see scan data constantly printing to console. Also when I set use_odometry = false, it was OK to build the initial map. Of course, once I start moving it around, the map become horrible.

So I am not sure what is causing the cartographer to not pick up the /scan data when odometry is used. The updated rqt_graph is shown below:

[image]

It appears to me that cartographer is correctly subscribed to /scan topic.

The github repo is also updated.

gaschler, 2019-01-07:
> It sounds like it never received the /scan topics. However, when I run rostopic echo /scan, I can see scan data constantly printing to console.

Please check what type the incoming messages are.
If it is "MultiEchoLaserScan", you need to set `options.num_multi_echo_laser_scans = 1`.
If it is "LaserScan", you need to set `options.num_laser_scans = 1` and so on.

I think the newest version of cartographer_ros should print a warning for these kind of mismatches.

YizhouGaoVictor, 2019-01-08:
Hi Gaschler,

I checked the message type, it is "LaserScan" from squash_depth_camera

And in the urdf file, it is set up as you described.

Is there any other possible cause?

Thanks.

JanezCim, 2020-03-18:
Did this get resolved?

MichaelGrupp, 2020-10-21:
Closing as inactive.

closed 2020-10-21 by MichaelGrupp