cartographer-project/cartographer_ros #1144: Problem with Tuning

lucabonamini, 2019-01-02:
Hello, 

I am having problems with cartographer's tuning.  I'd be very thankful for any help!

- here's the [rosbag_validate](https://gist.github.com/lucabonamini/6827597e22bf3231ea6f2b4d70a7be6c)

- all the [lua/lauch/urdf](https://github.com/lucabonamini/simulator) files 

- [bag](https://drive.google.com/open?id=1S8zbHgpgp_UuDi3lBjCOF56KPIbKQPt-)

Thank you!

lucabonamini, 2019-01-02:
Lidar works at 25 Hz. All sensors are "simulated" in UnReal. 

How can I tune laser SLAM? On which parameters should I work?

MichaelGrupp, 2020-10-23:
There is no specific advice I can give you because this depends very much on your custom setup. But you can have a look at the "tuning" sections of the documentation pages.

closed 2020-10-23 by MichaelGrupp