cartographer-project/cartographer_ros #1152: Experimental Velodyne VLP16 for 2D mapping

magiccjae, 2019-01-09:
My end goal is to generate indoor 3D map along with LiDAR point cloud. I was able to see 2D and 3D demos working, but not with my own "rosbag" files (2D and 3D both). So I think my first step would be to make 2D map working without IMU which is what I need some help on for now. 

1. Rosbag validate
[rosbag_validate_jae_test3.txt](https://gist.github.com/magiccjae/9677a5a600ddf0da019152e3747fb3f0#file-rosbag_validate_jae_test3-txt)

2. Git repository of the forked 'cartographer_ros'
[forked cartographer_ros](https://github.com/magiccjae/cartographer_ros)
Files that were used for 2D mapping:
cartographer_ros/configuration_files/my_2d.lua
cartographer_ros/launch/demo_my_2d.launch
cartographer_ros/launch/my_2d.launch
cartographer_ros/urdf/my_2d.urdf

3. Downloadable link to my Rosbag file
[jae_test3.bag](https://velodyne-my.sharepoint.com/:u:/p/jlee/EcwFrd6s7W9Klwrl_FNMf8IB3aIdx-ET8V3mAzJcxixOLA?e=fnK2C5)
This Rosbag file contains **/scan** topic messages. 

The problem I see in RVIZ is that **velodyne** frame goes all over the place with respect to **map** frame. Thus, cartographer seems not able to match the scans to generate map or trajectory. Please help me or guide me to where to start debugging. Thank you in advance and I look forward to the further discussion!

gaschler, 2019-01-09:
> E0108 16:14:07.830976 14540 rosbag_validate_main.cc:350] frame_id "velodyne" on topic scan has serialization time 1546986111.928834098 but sensor time 1546984955.325819000 differing by -1156.6 s.

Your timestamps in the velodyne topic are wrong.
As a temporary work around, you could set `options.use_pose_extrapolator = false` so the clock will be ignored and only the sensor header times are read.

magiccjae, 2019-01-10:
@gaschler ,

I think I fixed the timestamp issue by recording another rosbag file using proper **velodyne ros driver** and I don't see the error message about timing from `cartographer_rosbag_validate`.
New rosbag file can be found here:
[jae_test_2D.bag](https://velodyne-my.sharepoint.com/:u:/p/jlee/ERsxg6aSD-hDsluESStr0b0B4zORJLdvxnxiiT4yfQBsvQ?e=mMWpNw)

However, still I don't get any mapping activity going in RVIZ. When I launch 2D mapping, **velodyne frame** is not moving away from **odom** and **map frame**, thus not adding any **submap** and **trajectory**. 
The screenshot shows the result of mapping and it is easy to tell that laser scan had not been accumulated, but only shows the current scan of the environment.

[image]

What could I do differently to get mapping activity started? Any help or guidance would be appreciated!

Jaccica, 2019-01-22:
> @gaschler ,
> 
> I think I fixed the timestamp issue by recording another rosbag file using proper **velodyne ros driver** and I don't see the error message about timing from `cartographer_rosbag_validate`.
> New rosbag file can be found here:
> [jae_test_2D.bag](https://velodyne-my.sharepoint.com/:u:/p/jlee/ERsxg6aSD-hDsluESStr0b0B4zORJLdvxnxiiT4yfQBsvQ?e=mMWpNw)
> 
> However, still I don't get any mapping activity going in RVIZ. When I launch 2D mapping, **velodyne frame** is not moving away from **odom** and **map frame**, thus not adding any **submap** and **trajectory**.
> The screenshot shows the result of mapping and it is easy to tell that map had not been accumulated.
> 
> [image]
> 
> What could I do differently to get mapping activity started? Any help or guidance would be appreciated!

Can you solve your problem? I have the same problem.

magiccjae, 2019-01-29:
@Jaccica ,

I actually solved my problem. Please refer to the link to see what change was made in .lua file:
https://github.com/magiccjae/cartographer_ros/commit/e2b20a052de03f188e3d65e88932d7ccec14a14a

Also, another problem was in my rosbag file because **/scan** topic was not smoothly captured. When I looked **/scan** topic in my rosbag in RVIZ, it had a lot of jerky motion. So I captured new rosbag file on more stable platform so that LiDAR didn't move abruptly and 2D map is working great!

[image]

Thanks!

closed 2019-01-29 by magiccjae

joyde3382, 2019-03-28:
@magiccjae I'm having sort of the same issue that you had. When i try to run cartographer the map wont be displayed and i get a "Fixed Frame [map] does not exist" message in rviz. I tried using the same configuration as you have in your config file, just exchanging velodyne with rplidar (i am using an rplidar a1), but it's still not working. Any idea to what could be wrong? Have you experienced anyhting like that? 

also this is my rqt_graph:

[image]

Thanks in advance.

Rhonins, 2019-08-08:
> @magiccjae I'm having sort of the same issue that you had. When i try to run cartographer the map wont be displayed and i get a "Fixed Frame [map] does not exist" message in rviz. I tried using the [...]

Hello,
I am trying to do it with rplidar a2 but stuck in the same place. Did you figure that out?

Thanks.