cartographer-project/cartographer_ros #1153: Build cartographer_rviz independently

christqoh, 2019-01-09:
Background: I can not build cartographer_ros on my machine. This is ok as it is a custom build. I am able to build and run cartographer_ros in a docker. Visualization, however, should still take place directly on the machine.

Is there a way to only build cartographer_rviz (so I get the submaps plugin for rviz) without the need for building cartographer_ros or cartographer as building either will fail? I already am able to build cartographer_ros_msgs.

ojura, 2019-01-09:
No, because cartographer_rviz uses quite a bit from cartographer_ros/libcartographer (e.g. io functionality for texture unpacking).

christqoh, 2019-01-12:
It's not possible to build everything directly on the visualization machine. A workaround I found is to use 
```
sudo apt-get install ros-kinetic-cartographer*
```
on the host machine. In that version of cartographer, however, key features such as pure localization are not available yet. This keeps up the need for an up-to-date cartographer in a docker. 

Combining the two, one is able to get an outdated visualization to work on host while the latest cartographer is operating in a docker.

closed 2019-01-12 by christqoh

ojura, 2019-01-12:
I didn‘t mean to do the custom build exclusively on the visualization machine. If you can install the released Cartographer from repository, you should just as well be able build upstream Cartographer on your visualization machine, only for purposes of building cartographer_rviz. You can still do your custom build in Docker.