cartographer-project/cartographer_ros #1154: By not providing "FindAbseil.cmake" in CMAKE_MODULE_PATH this project has

jinfagang, 2019-01-12:
```
-- Detected gflags version: 2.2.1
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /usr with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Build type: Release
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at cartographer_ros/cartographer_ros/CMakeLists.txt:58 (find_package):
  By not providing "FindAbseil.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Abseil", but
  CMake did not find one.

  Could not find a package configuration file provided by "Abseil" with any
  of the following names:

    AbseilConfig.cmake
    abseil-config.cmake

  Add the installation prefix of "Abseil" to CMAKE_PREFIX_PATH or set
  "Abseil_DIR" to a directory containing one of the above files.  If "Abseil"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

```

Can not find Abseil...................

gaschler, 2019-01-14:
Closing as duplicate

closed 2019-01-14 by gaschler

vahidbehtajisiahkal, 2019-02-20:
Hi,
I have faced the same issue as above ( Could not find a package configuration file provided by "Abseil" with any of the following names:

    AbseilConfig.cmake
    abseil-config.cmake

  Add the installation prefix of "Abseil" to CMAKE_PREFIX_PATH or set
  "Abseil_DIR" to a directory containing one of the above files.  If "Abseil"
  provides a separate development package or SDK, be sure it has been
  installed)

 in Ubuntu 18.04 and melodic. Is there any solution ?