cartographer-project/cartographer_ros #1159: Unable to match scans (i.e. submap list is empty amd no map is being built)

ftbmynameis, 2019-01-15:
Hi Cartographer team,

I am trying to run cartographer on a turtlebot 2 with a 360° rplidar using only the laser scan (as of now) without odometry or an imu. I understand this might not be optimal, however I think my results shouldn't be that wrong even without odom/imu.
I have managed to run cartographer however it is not working and there has to be some major configuration issue because not a single submap can be matched. (in rviz if I disable "All" in submaps the list of all submaps contains only one element with the float in front of it increasing steadily). This also shows in the visualization because no map is built (i.e. I get only a "random" scattering of laser scan points in grey color while in other issues I have seen a "white map with black edges" is supposed to be built.

My rosbag validate result:
Is there a documentation for this tool available? I can barely make sense of the output data. It doesn't seem wrong, however I don't understand what the histogram/distribution is supposed to tell me.
https://gist.github.com/ftbmynameis/a6a51eba839f5906ecc971a7a4c99711

My branch with my configuration files: 
Note: I have installed cartographer_ros using the official manual (i.e. installed it in a catkin workspace) and made my configuration changes in /install_isolated/share/cartographer_ros. Since the official repository seems to be located in the "src" folder I have copied my configuration files there (in a new branch "turtlebot_config") and pushed them to my github fork of the repository. While doing so I have noticed that configuration files in the install_isolated/.. and src/.. folders do not match exactly which confuses me, but I have simply commited them hoping you guys have a better idea what is going on.
https://github.com/ftbmynameis/cartographer_ros/tree/turtlebot_config

My bag file is located here:
https://drive.google.com/open?id=1fs7C5IL_9VitraK0TFMOyb6UA18NkMbh

When running with the given configuration and bag file the point cloud in rviz starts sort of "spinning" which almost reminds me of a tumbling airplane going down and shows something major going wrong.

While I am here I also have a question about the parameter TRAJECTORY_BUILDER_nD.num_accumulated_range_data: as I understand this is a sensor dependent parameter (i.e. how many scans / how much data is provided by all my laser sensors) however I couldn't understand / figure out how to analyze my data to retrieve this value.

Thanks for your help and insights! If there is anything else required to be used I will try my best to provide it.

joekeo, 2019-04-26:
I am using the same lidar in a custom robot, initially only with lidar, I had issues finding which bits to tune and how to do it. A first tuning approach that produced good results, my config.lua was as follows:

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  use_odometry = false,
  provide_odom_frame = true,
  odom_frame = "odom",
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
--this one tries to match two laser scans together to estimate the position, 
--I think if not on it will rely more on wheel odometry
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
-- tune this value to the amount of samples (i think revolutions) to average over 
--before estimating te position of the walls and features in the environment
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1

--use or not use IMU, if used, the tracking_frame should be set to the one that the IMU is on
TRAJECTORY_BUILDER_2D.use_imu_data = false

--bandpass filter for lidar distance measurements
TRAJECTORY_BUILDER_2D.min_range = 0.3
TRAJECTORY_BUILDER_2D.max_range = 8.

--This is the scan matcher and the weights to different assumptions
--occupied_space gives more weight to the 'previous' features detected.
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.occupied_space_weight = 10.
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 10.
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 40.

return options

might not be optimal, but it is working the robot and its angular and linear speeds.

jonra1993, 2019-08-31:
Hi @joekeo I used you same configuration file without "use_pose_extrapolator" but list_map is empty and there is no map. Do you know which can be the problem?
[image]

MichaelGrupp, 2020-10-27:
Closing for inactivity, we can't invest time to help you with your setup anymore unfortunately.

closed 2020-10-27 by MichaelGrupp