cartographer-project/cartographer_ros #1161: Bag File Failing To Load

CycleMark, 2019-01-16:
Hi,

I am trying to get to a place where I can use Cartograpther for the map and localisation (as per 0 https://github.com/googlecartographer/cartographer_ros/issues/1131)

Still I cannot get it to load my bag file - I've read https://google-cartographer-ros.readthedocs.io/en/latest/your_bag.html  multiple times :-)

Any suggestions what is wrong with this:

mark@kobukiMxnet2:~/cartographer_v1.0$ roslaunch cartographer_turtlebot turtlebot_urg_lidar_2d.launch bag_filename:=/home/mark/cartographer_v1.0/cartographer_full_downstairs_20102018.bag

Output being:

... logging to /home/mark/.ros/log/ca8e2cf0-199e-11e9-ab2f-c1e19be8e044/roslaunch-kobukiMxnet2-4893.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.79:44836/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /urg_node/frame_id: plate_top_link

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    flat_world_imu_node (cartographer_turtlebot/cartographer_flat_world_imu_node)
    urg_node (urg_node/urg_node)

auto-starting new master
process[master]: started with pid [4905]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ca8e2cf0-199e-11e9-ab2f-c1e19be8e044
process[rosout-1]: started with pid [4925]
started core service [/rosout]
ERROR: cannot launch node of type [urg_node/urg_node]: urg_node
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/mark/cartographer_v1.0/src/cartographer_turtlebot/cartographer_turtlebot
ROS path [2]=/home/mark/cartographer_v1.0/install_isolated/share
ROS path [3]=/home/mark/ros_ws/src
ROS path [4]=/opt/ros/kinetic/share
process[cartographer_node-3]: started with pid [4935]
process[flat_world_imu_node-4]: started with pid [4943]
process[cartographer_occupancy_grid_node-5]: started with pid [4944]
[ INFO] [1547650549.307960887]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/src/cartographer_turtlebot/cartographer_turtlebot/configuration_files/turtlebot_urg_lidar_2d.lua' for 'turtlebot_urg_lidar_2d.lua'.
[ INFO] [1547650549.308924063]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1547650549.309065634]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1547650549.309296616]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1547650549.309414213]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1547650549.309866977]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1547650549.309986812]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1547650549.310188607]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1547650549.310301329]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1547650549.310689344]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1547650549.310808904]: I0116 14:55:49.000000  4935 configuration_file_resolver.cc:41] Found '/home/mark/cartographer_v1.0/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1547650549.353763059]: I0116 14:55:49.000000  4935 map_builder_bridge.cc:134] Added trajectory with ID '0'.
[ WARN] [1547650552.410724256]: W0116 14:55:52.000000  4935 node.cc:821] Expected topic "scan" (trajectory 0) (resolved topic "/scan") but no publisher is currently active.
[ WARN] [1547650552.411300367]: W0116 14:55:52.000000  4935 node.cc:821] Expected topic "odom" (trajectory 0) (resolved topic "/odom") but no publisher is currently active.
[ WARN] [1547650552.411818567]: W0116 14:55:52.000000  4935 node.cc:830] Currently available topics are: /constraint_list,/submap_list,/scan_matched_points2,/rosout,/tf,/imu,/rosout_agg,/map,/trajectory_node_list,/landmark_poses_list,

Many Thanks

Mark

Jaccica, 2019-01-22:
Hi, pay attention to the ERROR: cannot launch node of type [urg_node/urg_node]: urg_node you listed! Not bag file to load.

CycleMark, 2019-01-22:
@Jaccica Unfortunately that's not the problem. That is simply http://wiki.ros.org/urg_node which is unrelated.

MichaelGrupp, 2020-10-21:
Topic names must match what Cartographer expects. A topic remapping can help.

closed 2020-10-21 by MichaelGrupp