cartographer-project/cartographer_ros #1164: Cartographer doen not build map

Duke-Allen, 2019-01-17:
Hello, I'm trying to use cartographer to work with my own lidar, but the cartographer build map error.

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    cloud_node (rslidar_pointcloud/cloud_node)
    pointcloud_to_laserscan (pointcloud_to_laserscan/pointcloud_to_laserscan_node)
    rslidar_node (rslidar_driver/rslidar_node)

auto-starting new master
process[master]: started with pid [27758]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 00670e50-1a76-11e9-bb6b-00044ba57bd6
process[rosout-1]: started with pid [27771]
started core service [/rosout]
process[rslidar_node-2]: started with pid [27793]
process[cloud_node-3]: started with pid [27796]
process[pointcloud_to_laserscan-4]: started with pid [27804]
process[cartographer_node-5]: started with pid [27818]
process[cartographer_occupancy_grid_node-6]: started with pid [27829]
[ INFO] [1547742974.508969487]: publishing 84 packets per scan
[ INFO] [1547742974.542426613]: Cut at specific angle feature deactivated.
[ INFO] [1547742974.571653644]: Reconfigure Request
[ INFO] [1547742974.595145376]: Only accepting packets from IP address: 192.168.1.200
[ INFO] [1547742974.595241723]: Opening UDP socket: port 6699
[ INFO] [1547742974.606639878]: Only accepting packets from IP address: 192.168.1.200
[ INFO] [1547742974.606741217]: Opening UDP socket: port 7788
[ INFO] [1547742974.639566726]: Reconfigure Request
[ INFO] [1547742974.642201763]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer_ros/configuration_files/rslidar_scan.lua' for 'rslidar_scan.lua'.
[ INFO] [1547742974.643428102]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1547742974.643558208]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1547742974.643749430]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1547742974.643853809]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1547742974.644333465]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1547742974.645156753]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1547742974.645487488]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1547742974.645617146]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1547742974.646047972]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1547742974.646173822]: I0118 00:36:14.000000 27818 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1547742974.694199506]: I0118 00:36:14.000000 27818 submap_2d.cc:187] Added submap 1
[ INFO] [1547742974.694445894]: I0118 00:36:14.000000 27818 map_builder_bridge.cc:130] Added trajectory with ID '0'.
The distance resolution is 1cm
this->is_init_curve_ = true!
this->is_init_angle_ = true!
[ INFO] [1547742975.123070441]: I0118 00:36:15.000000 27818 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '621355968000000000'.
[ WARN] [1547742975.145505137]: W0118 00:36:15.000000 27818 pose_extrapolator.cc:168] Queue too short for velocity estimation. Queue duration: 0 ms
[ WARN] [1547742975.221921062]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.322125759]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.422971354]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.523732344]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.624511317]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.725444107]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.826108622]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742975.927110240]: W0118 00:36:15.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.027272285]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.128069562]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.229448443]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.330507308]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.431775986]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.531870482]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1547742976.632357663]: W0118 00:36:16.000000 27818 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[... cut ...]
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35
POSE_GRAPH.constraint_builder.min_score = 0.65

return options

Please help me, thanks

Duke-Allen, 2019-01-18:
I found the solution:
comment out the sentence
`<param name="/use_sim_time" value="true" />` in the file
but I don't know the reason,please give me the guidance

ojura, 2019-01-18:
@Duke-Allen, sorry for not being able to provide a more detailed explanation, but that is not a proper solution. If you are using a bag file or running a simulation in Gazebo, the proper thing to do is to use the simulation clock published on the `/clock` topic, which is controlled by setting `/use_sim_time` to true. If this is breaking your setup, this means that there is something wrong with your `/clock` topic. If you are playing a bag, be sure to use the `--clock` command-line parameter.

Duke-Allen, 2019-01-18:
@ojura , thank you very much for your guidance

closed 2019-01-20 by Duke-Allen