cartographer-project/cartographer_ros #1170: What determines the initial orientation of the map frame?

xielingsen, 2019-01-24:
I have a vehicle equipped with imu, odometry, lidar and I want to know the deviation of orientation of the IMU from the vehicle's main axis, which should be a fixed value. I drove the vehicle in a straight line and obtained this imu-vehicle deviation value by subtracting straight line direction from the imu orientation in map frame.
However, the deviation of the initial orientation of the map frame from the orientation of the imu frame( or lisar scan frame) does not seem to be fixed, so whenever I restart cartographer and perform a new measure , I get a different imu-vehicle deviation.
So, I want to know what determines the initial orientation of the map frame? Or is there a better way to get the imu-vehicle deviation?

NTUwilliam, 2019-03-07:
Hi all!

I think I got a similar issue. 
I'm trying to build a map and later do `pure_localization`. But when I started, the **orientation** of the `/map` frame is not correct. 
The correct **orientation** of the `/map` frame should be the same with the **orientation** of the `/base_link` frame (as shown below, the orientation deviation/error between `/map` frame and `/base_link` frame is around 30 degrees):
[image]

And the **`/tf`** between **`/map`** frame and **`/base_link`** frame is keep changing like this:

```
williamwoo@williamwoo:~$ rosrun tf tf_echo /map /base_link 
At time 1551951772.615
- Translation: [0.327, -0.494, -0.254]
- Rotation: in Quaternion [0.002, 0.007, -0.518, 0.856]
            in RPY (radian) [-0.004, 0.015, -1.088]
            in RPY (degree) [-0.222, 0.856, -62.336]
At time 1551951772.615
- Translation: [0.327, -0.494, -0.254]
- Rotation: in Quaternion [0.002, 0.007, -0.518, 0.856]
            in RPY (radian) [-0.004, 0.015, -1.088]
            in RPY (degree) [-0.222, 0.856, -62.336]
At time 1551951773.625
- Translation: [0.518, -0.821, -0.254]
- Rotation: in Quaternion [-0.000, 0.004, -0.518, 0.856]
            in RPY (radian) [-0.004, 0.007, -1.088]
            in RPY (degree) [-0.245, 0.399, -62.340]
At time 1551951774.625
- Translation: [0.681, -1.168, -0.254]
- Rotation: in Quaternion [-0.001, 0.005, -0.516, 0.857]
            in RPY (radian) [-0.006, 0.008, -1.083]
            in RPY (degree) [-0.372, 0.467, -62.079]
At time 1551951775.614
- Translation: [0.858, -1.494, -0.254]
- Rotation: in Quaternion [0.000, 0.003, -0.508, 0.861]
            in RPY (radian) [-0.003, 0.006, -1.067]
            in RPY (degree) [-0.152, 0.343, -61.110]
At time 1551951776.624
- Translation: [1.048, -1.834, -0.254]
- Rotation: in Quaternion [-0.002, 0.001, -0.524, 0.852]
            in RPY (radian) [-0.004, -0.001, -1.102]
            in RPY (degree) [-0.249, -0.058, -63.150]
At time 1551951777.619
- Translation: [1.215, -2.160, -0.254]
- Rotation: in Quaternion [0.000, 0.001, -0.514, 0.858]
            in RPY (radian) [-0.001, 0.002, -1.080]
            in RPY (degree) [-0.032, 0.142, -61.884]

```

The orientation of the **`/map`** frame is very critical for me because it's closely related to the `initial_pose` that I'm gonna [set for the `pure_localization` part](https://github.com/googlecartographer/cartographer_ros/issues/579#issuecomment-469160159).  

So I also want to know what determines the initial orientation of the **`/map`** frame? Any idea for solving this issue?

Thanks a lot !!
@ojura @gaschler @MichaelGrupp @damonkohler

IntelDan, 2019-06-20:
Any progress? I have the same problem. As the base_link's z is not zero, when I use load the pcd file and use mapserver map the 2D map, the lidar's z is zero, other frames's z are negative.
We use the lidar's position as original map frame when we start create map, is that right?