cartographer-project/cartographer_ros #1174: tf jumps a lot

Semih67, 2019-01-29:
I'm trying to build a 3d map. I have a tof camera. My porblem is that the base_link(i guess) is jumping a lot. So it rotates and changes it's position when a submap is published. And I also do not get a good map result:

> [ INFO] [1548760820.484216713]: I0129 12:20:20.000000 29712 submap_3d.cc:321] Added submap 1
> [ INFO] [1548760820.485049553]: I0129 12:20:20.000000 29712 map_builder_bridge.cc:130] Added trajectory with ID '0'.
> [ INFO] [1548760820.691109502]: Opened camera : PICOMONSTAR850GLASS
> BNO055 init complete
> [ INFO] [1548760822.000468632]: I0129 12:20:22.000000 29712 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636843576219399290'.
> [ INFO] [1548760822.000766575]: I0129 12:20:22.000000 29712 local_trajectory_builder_3d.cc:254] Extrapolator not yet initialized.
> [ INFO] [1548760825.228800242]: I0129 12:20:25.000000 29712 submap_3d.cc:321] Added submap 2
> [ INFO] [1548760829.233395669]: I0129 12:20:29.000000 29712 submap_3d.cc:321] Added submap 3
> [ INFO] [1548760830.046670119]: I0129 12:20:30.000000 29757 constraint_builder_3d.cc:296] Node (0, 10) with 31 points on submap (0, 0) differs by translation 0.08 rotation 0.006 with score 67.8%.
> [ INFO] [1548760830.046824771]: I0129 12:20:30.000000 29757 constraint_builder_3d.cc:311] 1 computations resulted in 1 additional constraints.
> Score histogram:
> Count: 1  Min: 0.678166  Max: 0.678166  Mean: 0.678166
> Rotational score histogram:
> Count: 1  Min: 0.939993  Max: 0.939993  Mean: 0.939993
> Low resolution score histogram:
> Count: 1  Min: 0.703455  Max: 0.703455  Mean: 0.703455
> [ INFO] [1548760830.053132655]: I0129 12:20:30.000000 29757 pose_graph_3d.cc:608] Translational residuals histogram:
> Count: 17  Min: 0.000000  Max: 0.049497  Mean: 0.013767
> [0.000000, 0.004950)                 #######    Count: 6 (35.294117%)    Total: 6 (35.294117%)
> [0.004950, 0.009899)                    ####    Count: 3 (17.647058%)    Total: 9 (52.941177%)
> [0.009899, 0.014849)                       #    Count: 1 (5.882353%)    Total: 10 (58.823528%)
> [0.014849, 0.019799)                      ##    Count: 2 (11.764706%)    Total: 12 (70.588234%)
> [0.019799, 0.024748)                    ####    Count: 3 (17.647058%)    Total: 15 (88.235291%)
> [0.024748, 0.029698)                            Count: 0 (0.000000%)    Total: 15 (88.235291%)
> [0.029698, 0.034648)                            Count: 0 (0.000000%)    Total: 15 (88.235291%)
> [0.034648, 0.039597)                            Count: 0 (0.000000%)    Total: 15 (88.235291%)
> [0.039597, 0.044547)                       #    Count: 1 (5.882353%)    Total: 16 (94.117645%)
> [0.044547, 0.049497]                       #    Count: 1 (5.882353%)    Total: 17 (100.000000%)
> [ INFO] [1548760830.053337641]: I0129 12:20:30.000000 29757 pose_graph_3d.cc:610] Rotational residuals histogram:
> Count: 17  Min: 0.000000  Max: 0.196431  Mean: 0.072938
> [0.000000, 0.019643)           #############    Count: 11 (64.705879%)    Total: 11 (64.705879%)
> [0.019643, 0.039286)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.039286, 0.058929)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.058929, 0.078572)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.078572, 0.098215)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.098215, 0.117859)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.117859, 0.137502)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.137502, 0.157145)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.157145, 0.176788)                            Count: 0 (0.000000%)    Total: 11 (64.705879%)
> [0.176788, 0.196431]                 #######    Count: 6 (35.294117%)    Total: 17 (100.000000%)
> [ INFO] [1548760832.429969455]: I0129 12:20:32.000000 29712 submap_3d.cc:321] Added submap 4
> [ INFO] [1548760835.602751146]: I0129 12:20:35.000000 29712 collated_trajectory_builder.cc:72] imu rate: 100.00 Hz 1.00e-02 s +/- 1.56e-03 s (pulsed at 97.12% real time)
> [ INFO] [1548760835.602905030]: I0129 12:20:35.000000 29712 collated_trajectory_builder.cc:72] odom rate: 14.25 Hz 7.02e-02 s +/- 5.47e-04 s (pulsed at 99.88% real time)
> [ INFO] [1548760835.603002243]: I0129 12:20:35.000000 29712 collated_trajectory_builder.cc:72] points2 rate: 2.50 Hz 4.00e-01 s +/- 2.08e-04 s (pulsed at 100.02% real time)
> [ INFO] [1548760836.030842281]: I0129 12:20:36.000000 29712 submap_3d.cc:321] Added submap 5
> [ INFO] [1548760837.241335977]: I0129 12:20:37.000000 29758 constraint_builder_3d.cc:311] 0 computations resulted in 0 additional constraints.
> Score histogram:
> Count: 1  Min: 0.678166  Max: 0.678166  Mean: 0.678166
> Rotational score histogram:
> Count: 1  Min: 0.939993  Max: 0.939993  Mean: 0.939993
> Low resolution score histogram:
> Count: 1  Min: 0.703455  Max: 0.703455  Mean: 0.703455
> [ INFO] [1548760837.253684557]: I0129 12:20:37.000000 29758 pose_graph_3d.cc:608] Translational residuals histogram:
> Count: 39  Min: 0.000250  Max: 0.061434  Mean: 0.011261
> [0.000250, 0.006368)            ############    Count: 24 (61.538460%)    Total: 24 (61.538460%)
> [0.006368, 0.012487)                       #    Count: 2 (5.128205%)    Total: 26 (66.666664%)
> [0.012487, 0.018605)                      ##    Count: 3 (7.692307%)    Total: 29 (74.358971%)
> [0.018605, 0.024724)                      ##    Count: 4 (10.256411%)    Total: 33 (84.615387%)
> [0.024724, 0.030842)                       #    Count: 1 (2.564103%)    Total: 34 (87.179489%)
> [0.030842, 0.036961)                       #    Count: 2 (5.128205%)    Total: 36 (92.307693%)
> [0.036961, 0.043079)                       #    Count: 1 (2.564103%)    Total: 37 (94.871796%)
> [0.043079, 0.049197)                       #    Count: 1 (2.564103%)    Total: 38 (97.435898%)
> [0.049197, 0.055316)                            Count: 0 (0.000000%)    Total: 38 (97.435898%)
> [0.055316, 0.061434]                       #    Count: 1 (2.564103%)    Total: 39 (100.000000%)
> [ INFO] [1548760837.253959942]: I0129 12:20:37.000000 29758 pose_graph_3d.cc:610] Rotational residuals histogram:
> Count: 39  Min: 0.005910  Max: 0.537817  Mean: 0.227491
> [0.005910, 0.059101)                   #####    Count: 10 (25.641026%)    Total: 10 (25.641026%)
> [0.059101, 0.112292)                       #    Count: 2 (5.128205%)    Total: 12 (30.769230%)
[... 266 lines cut ...]

POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher_3d.linear_xy_search_window = 4. --5.
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher_3d.linear_z_search_window = 0.8 --1.
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher_3d.angular_search_window = math.rad(11.) --15.

POSE_GRAPH.global_constraint_search_after_n_seconds = 20. --10.

--POSE_GRAPH.constraint_builder.ceres_scan_matcher_3d.ceres_solver_options.max_num_iterations = 5 --10

POSE_GRAPH.matcher_rotation_weight = 0.3
POSE_GRAPH.optimization_problem.huber_scale = 5e2 -- 5 * 10²
POSE_GRAPH.optimize_every_n_nodes = 10 --Global SLAM | Disable it with 0 --320
POSE_GRAPH.constraint_builder.sampling_ratio = .01 --.03
POSE_GRAPH.global_sampling_ratio =  0.001--0.003
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 9 --10
POSE_GRAPH.constraint_builder.min_score = 0.12 --0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options
```
my tf tree: 

[image]

And sometimes I get the warning: "queue too short for velocity" or "dropped range data"

gaschler, 2019-01-30:
Please post the text output of the rosbag_validate tool.

Semih67, 2019-02-04:
https://www.dropbox.com/s/fcg1epbj10m8k50/2019-02-04-08-35-31.bag?dl=0

This is the link to my bag file

closed 2019-02-19 by Semih67

Duke-Allen, 2019-03-03:
@Semih67 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)

Do you know the problem?