cartographer-project/cartographer_ros #1176: HasKey(key) Key 'trajectory_builder' not in dictionary:

LGK199408, 2019-01-30:
started roslaunch server http://192.168.1.111:33577/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)

auto-starting new master
process[master]: started with pid [6428]
ROS_MASTER_URI=http://192.168.1.111:11311

setting /run_id to d65363b6-2461-11e9-bda8-00bb6046080c
process[rosout-1]: started with pid [6441]
started core service [/rosout]
process[cartographer_node-2]: started with pid [6451]
[ INFO] [1548833824.409627718]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/src/cartographer_ros/cartographer_ros/configuration_files/revo_lds.lua' for 'revo_lds.lua'.
[ INFO] [1548833824.409887815]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1548833824.409938333]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1548833824.410003664]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1548833824.410049044]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1548833824.410180991]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1548833824.410228369]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1548833824.410352819]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1548833824.410378413]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1548833824.410501869]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1548833824.410530675]: I0130 15:37:04.000000  6451 configuration_file_resolver.cc:41] Found '/home/gdut/catkin_google_ws/devel_isolated/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
F0130 15:37:04.410740  6451 lua_parameter_dictionary.cc:399] Check failed: HasKey(key) Key 'trajectory_builder' not in dictionary:
{
  constant_odometry_rotational_variance = 0.000000,
  constant_odometry_translational_variance = 0.000000,
  horizontal_laser_max_range = 8.000000,
  horizontal_laser_min_range = 0.300000,
  horizontal_laser_missing_echo_ray_length = 0.000000,
  lookup_transform_timeout_sec = 0.200000,
  map_builder = {
    num_background_threads = 4.000000,
    pose_graph = {
      constraint_builder = {
        ceres_scan_matcher = {
          ceres_solver_options = {
            max_num_iterations = 10.000000,
            num_threads = 1.000000,
            use_nonmonotonic_steps = true,
          },
          occupied_space_weight = 20.000000,
          rotation_weight = 1.000000,
          translation_weight = 10.000000,
        },
        ceres_scan_matcher_3d = {
          ceres_solver_options = {
            max_num_iterations = 10.000000,
            num_threads = 1.000000,
            use_nonmonotonic_steps = false,
          },
          occupied_space_weight_0 = 5.000000,
          occupied_space_weight_1 = 30.000000,
          only_optimize_yaw = false,
          rotation_weight = 1.000000,
          translation_weight = 10.000000,
        },
        fast_correlative_scan_matcher = {
          angular_search_window = 0.523599,
          branch_and_bound_depth = 7.000000,
[... 158 lines cut ...]
  use_horizontal_multi_echo_laser = false,
  use_odometry_data = false,
}
*** Check failure stack trace: ***
    @     0x7fb89b3c85cd  google::LogMessage::Fail()
    @     0x7fb89b3ca433  google::LogMessage::SendToLog()
    @     0x7fb89b3c815b  google::LogMessage::Flush()
    @     0x7fb89b3cae1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x6a1b52  cartographer::common::LuaParameterDictionary::CheckHasKey()
    @           0x6a1bae  cartographer::common::LuaParameterDictionary::CheckHasKeyAndReference()
    @           0x6a1f2e  cartographer::common::LuaParameterDictionary::GetDictionary()
    @           0x5c8da1  cartographer_ros::CreateTrajectoryOptions()
    @           0x5821bc  cartographer_ros::LoadOptions()
    @           0x57d24b  cartographer_ros::(anonymous namespace)::Run()
    @           0x57a994  main
    @     0x7fb897082830  __libc_start_main
    @           0x57cf69  _start
    @              (nil)  (unknown)
[cartographer_node-2] process has died [pid 6451, exit code -6, cmd /home/gdut/catkin_google_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_node -configuration_directory /home/gdut/catkin_google_ws/src/cartographer_ros/cartographer_ros/configuration_files -configuration_basename revo_lds.lua scan:=scan __name:=cartographer_node __log:=/home/gdut/.ros/log/d65363b6-2461-11e9-bda8-00bb6046080c/cartographer_node-2.log].
log file: /home/gdut/.ros/log/d65363b6-2461-11e9-bda8-00bb6046080c/cartographer_node-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MichaelGrupp, 2019-02-04:
`use_constant_odometry_variance`

What's that?

LGK199408, 2019-02-21:
`use_constant_odometry_variance`

What's that? This doesn't parameter doesn't exist in Cartographer, so I assume your config has errors.

in "revo_lds_.lua"

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
map_builder = MAP_BUILDER,
trajectory_builder = TRAJECTORY_BUILDER,
map_frame = "map",
tracking_frame = "laser",
published_frame = "laser",
odom_frame = "odom",
provide_odom_frame = true,
use_odometry = false,
num_laser_scans = 1,
num_multi_echo_laser_scans = 0,
num_subdivisions_per_laser_scan = 1,
num_point_clouds = 0,
lookup_transform_timeout_sec = 0.2,
submap_publish_period_sec = 0.3,
pose_publish_period_sec = 5e-3,
trajectory_publish_period_sec = 30e-3,
rangefinder_sampling_ratio = 1.,
odometry_sampling_ratio = 1.,
imu_sampling_ratio = 1.,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 35
TRAJECTORY_BUILDER_2D.min_range = 0.3
TRAJECTORY_BUILDER_2D.max_range = 8.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1
POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35
POSE_GRAPH.constraint_builder.min_score = 0.65
return options

and in "demo_revo_lds_rplidar.launch"

LGK199408, 2019-02-21:
<launch>
  <param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename revo_lds_rplidar.lua"
      output="screen">
    <remap from="scan" to="scan" />
  </node>
  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
</launch>

LGK199408, 2019-02-21:
cartographer_ros--cartographer_ros--launch--demo_revo_lds.launch

<launch> 
  <param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename revo_lds.lua"
      output="screen">
    <remap from="scan" to="scan" />
  </node>
</launch>

cartographer_ros--cartographer_ros--configuration_files--revo_lds.lua
include "map_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  sensor_bridge = {
    horizontal_laser_min_range = 0.3,
    horizontal_laser_max_range = 8,
    horizontal_laser_missing_echo_ray_length = 1.2,
    constant_odometry_translational_variance = 0.,
    constant_odometry_rotational_variance = 0.,
  },
  map_frame = "map",
  tracking_frame = "laser",
  published_frame = "laser",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_constant_odometry_variance = true,
  constant_odometry_translational_variance = 1e-2,
  constant_odometry_rotational_variance = 1e-1,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.1,
  horizontal_laser_max_range = 30.,
  horizontal_laser_missing_echo_ray_length = 5.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
POSE_GRAPH.optimization_problem.huber_scale = 1e2
return options

LGK199408, 2019-02-21:
https://blog.csdn.net/ywj447/article/details/52922487

MichaelGrupp, 2019-02-21:
Some of these options have been removed quite a while ago (https://github.com/googlecartographer/cartographer_ros/commit/ca5b7d1bff9c865a70dff2386c7df7f26c33921f), so I assume that either you use an old version of Cartographer or you use an old configuration that is not compatible with the new Cartographer version.

LGK199408, 2019-02-21:
I git the new Cartographer in Github,how can I use the new configuration,I use the rplidarA1

MichaelGrupp, 2020-10-26:
I would recommend to start with the configuration of the backpack demo examples, because they should work. And then add your changes. See also https://google-cartographer-ros.readthedocs.io/en/latest/your_bag.html#create-a-lua-configuration

closed 2020-10-26 by MichaelGrupp