cartographer-project/cartographer_ros #1178: After a while, processing gets very slow.

dan9thsense, 2019-01-30:
I have both 2d and 3d cartographer working great for localization in small and medium size environments.  However, when I try to explore a large environment, the processing eventually (after 30 minutes or so) slows down and grinds to a halt.

I am wondering if I need to close submaps or something like that as I go along.  If so, how do I do that and also, how do I save the overall localization?

dan9thsense, 2019-02-04:
Thanks, I'll try that.  What I meant was, if I get rid of old submaps, will I still have global localization?

MichaelGrupp, 2019-02-04:
(I assume you have a "frozen" background map in which you want to localize)

dan9thsense, 2019-02-05:
I am trying to create a 3d map while exploring a very large area (2 x 2 km) and remain localized with respect to the original origin throughout the process.  In previous projects, I have done that over smaller areas (.1 km x .1 km) and then saved the map at the end and also created a 2d ROS map from that.  In this case, I don't need the ROS map (although it would be nice to have), but I do need to stay localized  while exploring.

MichaelGrupp, 2019-02-08:
Ah, I see. So you do live mapping, I thought you meant pure localization.

Changing some parameters can already have a positive effect on runtime efficiency. For example disabling global constraint search by setting `POSE_GRAPH.global_constraint_search_after_n_seconds` to very high value, lowering the `POSE_GRAPH.constraint_builder.sampling_ratio`, making the min_scores higher to let the branch-and-bound search exit earlier, ...

dan9thsense, 2019-02-10:
Thanks for the suggestions.  I was not able to find guidelines for values of those parameters, so I just tried plugging in `POSE_GRAPH.global_constraint_search_after_n_seconds = 100` and `POSE_GRAPH.constraint_builder.sampling_ratio = 0.05` and the efficiency did improve.  It's hard to figure out the impact of those changes though.

MichaelGrupp, 2019-02-11:
Sounds like an example for these parameters would be a useful addition to the [tuning guide](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html).

MichaelGrupp, 2019-02-11:
Actually, it's already there: https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency

dan9thsense, 2019-02-11:
Yes, that is a very helpful list.  Is there someplace where the default values for all parameters are listed?  I found some values by using grep, but there must be a better way.

MichaelGrupp, 2019-02-11:
I actually have a pull request in `cartographer` for an utility that tries to help with exactly this :smile: 

https://github.com/googlecartographer/cartographer/pull/1508

It's not merged yet, but maybe you can upvote it

dan9thsense, 2019-02-14:
upvoted!

closed 2020-10-23 by MichaelGrupp