cartographer-project/cartographer_ros #1179: Actual 3D pointcloud creation

jattiogbe, 2019-02-04:
Hi,

I am new to Cartographer, I've tested the demos and I was disappointed to see what they call 3D SLAM since I was expecting to see the pointcloud of an actual 3D museum, not just the floor...

I've read lots of issues here and I don't understand all of them. I think the closer ones to my goal are  [this one ](https://github.com/googlecartographer/cartographer_ros/issues/1168) and [this one](https://github.com/googlecartographer/cartographer_ros/issues/1049).
I don't understand how to process things explained here.
My goal is to get a .ply file out of my lidar data (ie bagfiles), I am not "fluent" in this developping environment words : asset, pipeline.
Can you explain me what should I do ? what should I precisely type, execute ?

thanks.

MichaelGrupp, 2019-02-05:
duplicate: https://github.com/googlecartographer/cartographer/issues/1515

Please only keep one issue open, duplication doesn't help anyone ;)

jattiogbe, 2019-02-05:
Yes you're right ;)

closed 2019-02-05 by jattiogbe