cartographer-project/cartographer_ros #1182: 2D using IMU - expected input

christqoh, 2019-02-07:
Which frame of reference does cartographer expect for the `/imu` topic? In my setup, I use the [ROS standard REP 103](http://www.ros.org/reps/rep-0103.html#axis-orientation) orientation with `[x,y,z]`pointing in [forward, left, up] directions. This, at rest, gives accelerations `[a_x, a_y, a_z] = [0, 0, -g]` which seems physically reasonable.

However, this flips my map. Not really sure why this is happening. Flipping the sign of `a_z` fixes this but a positiv earth acceleration in a coordinate system where `z-axis` is pointing up does not really make sense. (Also, this gives a left-oriented coordinate system)

Can you please elaborate on the expected IMU axis of cartographer? Cheers!

- The according urdf file is located [here](https://github.com/Roboy/cartographer_ros/blob/roboy/cartographer_ros/urdf/roboy.urdf). You can run the code using [this `launch`-file](https://github.com/Roboy/cartographer_ros/blob/roboy/cartographer_ros/launch/roboy_indoor_offline.launch).
- You can find the data [here](https://drive.google.com/drive/folders/1RGpb66U0dW__ybrM4_9pXSbtU3ZyHtGO). Please use the `2019-02-07-16-10-08.bag`.

precyon, 2019-02-13:
If you follow REP 103 and the robot is stationary with its Z-axis pointed upwards, then it will read `[0, 0, +g]`.  Think about this: if the IMU is freely falling it's acceleration is `-g` but the accelerometer will read `[0, 0, 0]`.  

Have a look at this section in REP 0145: http://www.ros.org/reps/rep-0145.html#data-sources

christqoh, 2019-02-15:
Thank you very much @precyon, I assume the same is true for x and y pointing upwards.

closed 2019-02-15 by christqoh