cartographer-project/cartographer_ros #1184: Begginer Questions: How to check if loop closure occured? Which map to use?

fprott, 2019-02-10:
Hello,

First of all thank you for looking into my beginner question.

I recently started working with cartographer and noticed that the documentation, while insightful at many points, left me with some questions that I can not answer. First of all I read the documentation (but I am in no way someone who is able to recall everything) and I actively tried out most features with the examples (incidentally found a mistakes in the documentation which I already fixed).

Now I have a few burning questions (which I would like to add to the documentation):
-What is the best way to recognize that a loop-closure has happened? Is there a flag that I am somehow missing?
-Am I correct in the assumption that one should prefer to use the finished submaps and not the map provided by the cartographer_occupancy_grid_node? 
-Is there a way to map landmarks (id's) onto the map or do I have to do that myself with the landmarks topic published by the cartographer_node?

Thank you for your help. I am very sorry if I missed something but I have been looking for a full day as of now and I don't seem to get it.

MichaelGrupp, 2019-02-11:
> Am I correct in the assumption that one should prefer to use the finished submaps and not the map provided by the cartographer_occupancy_grid_node?

What do you mean with "use"? The `cartographer_occupancy_grid_node` only converts and publishes the submaps of Cartographer to the ROS map format, also the unfinished ones. This might be useful if you have another node that needs the live map in this format. 

> landmarks

If you configured Cartographer to use landmarks, you need to publish [weighted landmark poses relative to the tracking frame](https://google-cartographer.readthedocs.io/en/latest/cost_functions.html#landmark-cost-function). How you get these landmark poses is up to you, they can come e.g. from a QR code detector.

This thread already had an intense discussion about landmarks and might contain useful information: https://github.com/googlecartographer/cartographer_ros/issues/1067

I also wrote a [script](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/scripts/dev/publish_fake_random_landmarks.py) to simplify testing with dummy landmarks. You can run it in parallel to Cartographer and you should be able to see landmark markers in RViz. You can also customize it and integrate it into your code to publish real landmarks.

Sumyer, 2019-11-25:
@MichaelGrupp 
Thank you for your contribution!
But what about the answer to the first question?
How to identify if a loop closure is detected? I had a doubt when using the cartographer. There is a loop closure in my map, but I am sorry, I can't tell if the cartographer recognizes it. I wonder if you have any suggestions?

MichaelGrupp, 2020-10-23:
There is currently no specific interface in ROS to get callbacks when loop closures happened AFAIK.

closed 2020-10-23 by MichaelGrupp