cartographer-project/cartographer_ros #1186: SLAM through elevators

xperroni, 2019-02-11:
I'm looking into the problem of performing SLAM in a multi-floor environment where the robot travels across floors by elevator. I've thought of three possible approaches:

1. Map each floor separately in 2D mode, restarting Cartographer with the appropriate map each time the robot changes floors;
2. Keep a single 3D map, using `/finish_trajectory` / `cartographer_start_trajectory` to shift the robot vertically during mapping / localization to account for floor changes;
3. Keep a single 3D map, and somehow communicate the robot's vertical movement to Cartographer (e.g. by manipulating the Z coordinate in odometry, or perhaps the TF transforms).

For option (1), is there a better way to switch among maps than shutting down then restarting `cartographer_node` (and possibly other nodes that depend on it)?

For option (2), I'm just assuming it's possible to arbitrarily change the current robot location during mapping and that Cartographer will do The Right Thing ®, take the new location at face value and continue mapping from there. Is that a reasonable assumption?

For option (3), how easy is it to "fool" Cartographer into shifting the robot's location up / down, assuming this is at all possible?

Semih67, 2019-02-18:
I think the map will be drawn correct if you use an imu and the 3d mode

MichaelGrupp, 2020-10-17:
1) and 2) sound like reasonable approaches, 3) is probably unrealistic because you can't measure the elevator accurately.

closed 2020-10-17 by MichaelGrupp