cartographer-project/cartographer_ros #1187: cartographer node keeps crashing with the demo bags

kapz28, 2019-02-12:
Link to rosbag validate output:
https://gist.github.com/kapz28/aac27da5585907ee3f41a67ebef22e23

I utilized the ros cartographer setup tutorial 
https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#

link to the bag file = wget -P ~/Downloads https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/cartographer_paper_deutsches_museum.bag

As per the google cartographer Tutorial when I run the commands
1. source install_isolated/setup.bash
.... 
2. roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag
.....

This is the following output

... logging to /home/kapz28/.ros/log/8366507a-2e55-11e9-a5c0-1062e5e8a405/roslaunch-LAPTOP-ET54V7S9-149.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://LAPTOP-ET54V7S9:64344/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [159]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8366507a-2e55-11e9-a5c0-1062e5e8a405
process[rosout-1]: started with pid [170]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [173]
process[cartographer_node-3]: started with pid [178]
process[cartographer_occupancy_grid_node-4]: started with pid [185]
[ INFO] [1549928046.217446500]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1549928046.221234200]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1549928046.222133000]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1549928046.224033100]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1549928046.225203900]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1549928046.238719000]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1549928046.239859600]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1549928046.241598200]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1549928046.242611900]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1549928046.244416400]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1549928046.245551300]: I0211 18:34:06.000000   178 configuration_file_resolver.cc:41] Found '/home/kapz28/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
F0211 18:34:06.259860   193 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
[FATAL] [1549928046.269506100]: F0211 18:34:06.000000   193 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
F0211 18:34:06.260058   194 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1) F0211 18:34:06.260207   195 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1) F0211 18:34:06.260440   196 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
[FATAL] [1549928046.278696500]: F0211 18:34:06.000000   194 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
F0211 18:34:06.260058   194 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1) F0211 18:34:06.260207   195 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1) F0211 18:34:06.260440   196 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
[FATAL] [1549928046.280854300]: F0211 18:34:06.000000   195 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
F0211 18:34:06.260058   194 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1) F0211 18:34:06.260207   195 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1) F0211 18:34:06.260440   196 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
[FATAL] [1549928046.282213400]: F0211 18:34:06.000000   196 thread_pool.cc:85] Check failed: nice(10) != -1 (-1 vs. -1)
[ INFO] [1549928046.283207800]: I0211 18:34:06.000000   178 submap_2d.cc:187] Added submap 1
process[rviz-5]: started with pid [192]
process[playbag-6]: started with pid [204]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-kapz28'
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
*** Check failure stack trace: ***
    @     0x7fd1829cc0cd  google::LogMessage::Fail()
*** Check failure stack trace: ***
*** Check failure stack trace: ***
*** Check failure stack trace: ***
[ INFO] [1549928047.284614600]: I0211 18:34:07.000000   178 map_builder_bridge.cc:130] Added trajectory with ID '0'.
    @     0x7fd1829cdf33  google::LogMessage::SendToLog()
    @     0x7fd1829cc0cd  google::LogMessage::Fail()
    @     0x7fd1829cc0cd  google::LogMessage::Fail()
    @     0x7fd1829cc0cd  google::LogMessage::Fail()
    @     0x7fd1829cbc28  google::LogMessage::Flush()
    @     0x7fd1829cdf33  google::LogMessage::SendToLog()
    @     0x7fd1829cdf33  google::LogMessage::SendToLog()
    @     0x7fd1829cdf33  google::LogMessage::SendToLog()
    @     0x7fd1829ce999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fd1829cbc28  google::LogMessage::Flush()
    @     0x7fd1829cbc28  google::LogMessage::Flush()
    @     0x7fd1829cbc28  google::LogMessage::Flush()
    @     0x7fd1834623ed  cartographer::common::ThreadPool::DoWork()
    @     0x7fd1829ce999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fd1829ce999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fd1829ce999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fd17f51d57f  (unknown)
    @     0x7fd1834623ed  cartographer::common::ThreadPool::DoWork()
    @     0x7fd1834623ed  cartographer::common::ThreadPool::DoWork()
    @     0x7fd1834623ed  cartographer::common::ThreadPool::DoWork()
    @     0x7fd1811076db  start_thread
    @     0x7fd17f51d57f  (unknown)
    @     0x7fd17f51d57f  (unknown)
    @     0x7fd17ebc188f  clone
    @     0x7fd17f51d57f  (unknown)
[cartographer_node-3] process has died [pid 178, exit code -6, cmd /home/kapz28/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/kapz28/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/kapz28/.ros/log/8366507a-2e55-11e9-a5c0-1062e5e8a405/cartographer_node-3.log].
log file: /home/kapz28/.ros/log/8366507a-2e55-11e9-a5c0-1062e5e8a405/cartographer_node-3*.log

kapz28, 2019-02-12:
I run it now and this is the new output I get 
W0211 19:21:16.640622   464 rosbag_validate_main.cc:166] Sensor with frame_id "vertical_laser_link" measurements overlap in time. Previous range message, ending at time stamp 635682439310086097, must finish before current range message, which ranges from 635682439310071250 to 635682439310258056
W0211 19:23:24.175415   464 rosbag_validate_main.cc:203] Sensor with frame_id "vertical_laser_link" range measurements have longest overlap of 0.0014847 s
I0211 19:23:24.178586   464 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "horizontal_laser_2d" (frame_id: "horizontal_laser_link"):
Count: 70357  Min: 0.019381  Max: 0.041240  Mean: 0.027189
[0.019381, 0.021567)                            Count: 11 (0.015635%)   Total: 11 (0.015635%)
[0.021567, 0.023753)                            Count: 89 (0.126498%)   Total: 100 (0.142132%)
[0.023753, 0.025939)                  ######    Count: 22294 (31.686968%)       Total: 22394 (31.829100%)
[0.025939, 0.028125)             ###########    Count: 37741 (53.642139%)       Total: 60135 (85.471237%)
[0.028125, 0.030311)                       #    Count: 3565 (5.067015%) Total: 63700 (90.538254%)
[0.030311, 0.032497)                            Count: 466 (0.662336%)  Total: 64166 (91.200592%)
[0.032497, 0.034682)                            Count: 935 (1.328937%)  Total: 65101 (92.529526%)
[0.034682, 0.036868)                       #    Count: 5123 (7.281436%) Total: 70224 (99.810966%)
[0.036868, 0.039054)                            Count: 120 (0.170559%)  Total: 70344 (99.981522%)
[0.039054, 0.041240]                            Count: 13 (0.018477%)   Total: 70357 (100.000000%)
I0211 19:23:24.227489   464 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "imu" (frame_id: "imu_link"):
Count: 478243  Min: 0.000004  Max: 0.022381  Mean: 0.003996
[0.000004, 0.002242)                 #######    Count: 169318 (35.404179%)      Total: 169318 (35.404179%)
[0.002242, 0.004479)                            Count: 4804 (1.004510%) Total: 174122 (36.408688%)
[0.004479, 0.006717)            ############    Count: 283642 (59.309181%)      Total: 457764 (95.717865%)
[0.006717, 0.008955)                            Count: 10539 (2.203691%)        Total: 468303 (97.921562%)
[0.008955, 0.011192)                            Count: 1495 (0.312603%) Total: 469798 (98.234161%)
[0.011192, 0.013430)                            Count: 7646 (1.598769%) Total: 477444 (99.832932%)
[0.013430, 0.015668)                            Count: 563 (0.117723%)  Total: 478007 (99.950653%)
[0.015668, 0.017905)                            Count: 197 (0.041192%)  Total: 478204 (99.991844%)
[0.017905, 0.020143)                            Count: 27 (0.005646%)   Total: 478231 (99.997490%)
[0.020143, 0.022381]                            Count: 12 (0.002509%)   Total: 478243 (100.000000%)
I0211 19:23:24.279162   464 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "vertical_laser_2d" (frame_id: "vertical_laser_link"):
Count: 69362  Min: 0.017196  Max: 0.040862  Mean: 0.027579
[0.017196, 0.019562)                            Count: 1 (0.001442%)    Total: 1 (0.001442%)
[0.019562, 0.021929)                            Count: 8 (0.011534%)    Total: 9 (0.012975%)
[0.021929, 0.024296)                            Count: 93 (0.134079%)   Total: 102 (0.147055%)
[0.024296, 0.026662)              ##########    Count: 33182 (47.838875%)       Total: 33284 (47.985928%)
[0.026662, 0.029029)                ########    Count: 27494 (39.638420%)       Total: 60778 (87.624344%)
[0.029029, 0.031395)                            Count: 1218 (1.756005%) Total: 61996 (89.380356%)
[0.031395, 0.033762)                            Count: 179 (0.258066%)  Total: 62175 (89.638420%)
[0.033762, 0.036129)                      ##    Count: 6652 (9.590265%) Total: 68827 (99.228683%)
[0.036129, 0.038495)                            Count: 501 (0.722297%)  Total: 69328 (99.950981%)
[0.038495, 0.040862]                            Count: 34 (0.049018%)   Total: 69362 (100.000000%)

Lingbo2017, 2019-03-18:
Hello! I have met the same problem with you, and my computer configuration is here:
**ubuntu 16.04, 64bit , kinetic.**

All of them went well in compiling process until I launch the 2D demo using this command:
**roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag**

The ERROR part came as follows:
**[ INFO] [1552891680.239283186, 1432647016.574284759]: I0318 14:48:00.000000 14323 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '635682438164904058'.
terminate called after throwing an instance of 'std::out_of_range'
what(): map::at
[cartographer_node-3] process has died [pid 14323, exit code -6, cmd /home/lingbo/catkin_cartographer/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/lingbo/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/lingbo/.ros/log/c41d31ce-4949-11e9-8afe-54271ed5aac5/cartographer_node-3.log].
log file: /home/lingbo/.ros/log/c41d31ce-4949-11e9-8afe-54271ed5aac5/cartographer_node-3*.log**

Some has inferred that this is an alignment issue from running on [32-bit.](https://github.com/googlecartographer/cartographer_ros/issues/48)
**But I am pretty sure that both my cpu and ubuntu 16.04 OS are 64 bit.**

So could anyone help me figure this out here? Thank you very much!

kimtj5521, 2019-04-11:
@Lingbo2017 @kapz28 
Hi. I have met the same problem.
I had wasted a lot of time because of this strange error.

Finally, I found a solution.

The situation where I experienced this error is as follows.

1. Install Ubuntu 16.04, 64bit / (i7 cpu, 16GB RAM, enough SSD, laptop, desktop, intel NUC, et al..)
2. sudo apt-get update & upgrade
3. Install Qt 5.x -> VTK -> PCL -> OpenCV 3.x -> ROS kinetic -> cartographer
  (VTK optional, If you have Nvidia graphic card or some other graphic card, the graphic card drive does not matter.)

I always met that error when I tried cartographer after installing it like above.

I don't know the exact cause of the error.
However, this method solved the error for me.

1. Install Ubuntu 16.04, 64 bit.
2. sudo apt-get update & upgrade.
3. **ROS kinetic INSTALL FIRST!!** (If you want to install other ros distro, you can try. I think it doesn't matter. I'm just ROS kinetic user.)
4. intsall cartographer.
5. And then install what you want to use any thing.

I think this seems to be ralated to the version of the library used by ROS kinetic.

I hoped this method will help you.

( p.s. If you use this way, you have to reinstall ubuntu. sorry...)

closed 2019-04-12 by kapz28