cartographer-project/cartographer_ros #1188: Rotating VLP-16

athtest800, 2019-02-12:
Hi!
Do you think a rotating VLP-16 will work, in order to cover wider FOV?
Will it be better to rigid mount the IMU with the lidar so both are spinning?
Or is is better to use encoder for spinning lidar and continuously translate rotation to IMU.
Lastly, the num_accumulated_range_data will be the scans for one VLP-16 rotation? eg. 1 sec?
Please, give me your recommendations and experience.
Thank you.

precyon, 2019-02-12:
@athtest800, what is your application? Do you want 2D or 3D maps? Are all your scans going to be horizontal?

IMUs are mounted rigidly on your robot and the URDF is populated with the locations of the lidar and the IMU with respect to a reference point on the robot.

`num_accumulated_range_data` depends on how you get your pointcloud. If you are using the VLP-16 and the popular Velodyne driver for ROS, it accumulates the point cloud for you. So try with,

`num_accumulated_range_data = 1`

closed 2019-02-13 by athtest800

Duke-Allen, 2019-03-03:
@precyon 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
This is my pointcloud map, it looks like lapped:
[image]
bag_xray_xy_all:
[image]
bag_xray_xz_all:
[image]
bag_xray_yz_all:
[image]
Do you know the problem?

athtest800, 2019-03-07:
Hi!
I am interested for your .bag file.
Could you upload it in another host?
Thanks

Duke-Allen, 2019-03-07:
@athtest800 
I am sorry that I could not use google drive and dropbox. Could you open this [link](https://share.babel.cc/share.do?s=734993018329350)? 
the account is 15838868117
password is 123456789

athtest800, 2019-03-08:
The imu has angular velocity in its data. It means that it rotates.
So, I think that your imu gives wrong data.
Also, could you give an indoor sample of the robosense?
I would like to test something.
Thank you.

Duke-Allen, 2019-03-08:
@athtest800 
Thanks for your reply.
My IMU is MPU6050. Why does it give wrong data?
Do you mean you need a data bag of the lidar?

Duke-Allen, 2019-03-09:
[This](https://share.babel.cc/share.do?s=2578997252785691) is the indoor pointcloud data of lidar.

Duke-Allen, 2019-03-09:
@athtest800 
Hi, I filter the IMU data and the validate result is:

> 
W0309 23:56:29.330384  4806 rosbag_validate_main.cc:103] frame_id imu time 1552117074010934920: IMU linear acceleration is 2.38753 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0-1.08145 0.0394775 0002.1282
W0309 23:56:29.668264  4806 rosbag_validate_main.cc:103] frame_id imu time 1552117082020941448: IMU linear acceleration is 2.96425 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.153125 01.52886 02.53494
W0309 23:56:30.536268  4806 rosbag_validate_main.cc:103] frame_id imu time 1552117101942092808: IMU linear acceleration is 2.95918 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -1.08384 02.28611 01.53484
I0309 23:56:33.832828  4806 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 16577  Min: 0.006398  Max: 0.013598  Mean: 0.010000
[0.006398, 0.007118)	                    	Count: 1 (0.006032%)	Total: 1 (0.006032%)
[0.007118, 0.007838)	                    	Count: 1 (0.006032%)	Total: 2 (0.012065%)
[0.007838, 0.008558)	                    	Count: 92 (0.554986%)	Total: 94 (0.567051%)
[0.008558, 0.009278)	                    	Count: 17 (0.102552%)	Total: 111 (0.669602%)
[0.009278, 0.009998)	              ######	Count: 5381 (32.460640%)	Total: 5492 (33.130241%)
[0.009998, 0.010718)	       #############	Count: 10974 (66.200157%)	Total: 16466 (99.330399%)
[0.010718, 0.011438)	                    	Count: 21 (0.126682%)	Total: 16487 (99.457077%)
[0.011438, 0.012158)	                    	Count: 86 (0.518791%)	Total: 16573 (99.975868%)
[0.012158, 0.012878)	                    	Count: 3 (0.018097%)	Total: 16576 (99.993965%)
[0.012878, 0.013598]	                    	Count: 1 (0.006032%)	Total: 16577 (100.000000%)
E0309 23:56:33.833559  4806 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101545 s, recommended is [0.0005, 0.05] s with no jitter.
I0309 23:56:33.833832  4806 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 1644  Min: 0.100073  Max: 0.101545  Mean: 0.100801
[0.100073, 0.100220)	                    	Count: 1 (0.060827%)	Total: 1 (0.060827%)
[0.100220, 0.100367)	                    	Count: 0 (0.000000%)	Total: 1 (0.060827%)
[0.100367, 0.100515)	                    	Count: 0 (0.000000%)	Total: 1 (0.060827%)
[0.100515, 0.100662)	                    	Count: 17 (1.034063%)	Total: 18 (1.094890%)
[0.100662, 0.100809)	      ##############	Count: 1141 (69.403893%)	Total: 1159 (70.498787%)
[0.100809, 0.100956)	              ######	Count: 472 (28.710463%)	Total: 1631 (99.209244%)
[0.100956, 0.101103)	                    	Count: 12 (0.729927%)	Total: 1643 (99.939171%)
[0.101103, 0.101251)	                    	Count: 0 (0.000000%)	Total: 1643 (99.939171%)
[0.101251, 0.101398)	                    	Count: 0 (0.000000%)	Total: 1643 (99.939171%)
[0.101398, 0.101545]	                    	Count: 1 (0.060827%)	Total: 1644 (100.000000%)

The map is also messy. Do you know the problem?

Duke-Allen, 2019-03-15:
@athtest800 
I realized that the reason why the car was rotating was because there was some angular velocity on the x and y axis, but the car is in the horizontal movement. So I set the angular velocity zero and my car does not rotate. But the track differs from the real movement. 
[The new data bag:](https://share.babel.cc/share.do?s=4553105679638233)

> I0315 15:39:30.105394 10492 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 18056  Min: 0.004353  Max: 0.013893  Mean: 0.010000
[0.004353, 0.005307)	                    	Count: 1 (0.005538%)	Total: 1 (0.005538%)
[0.005307, 0.006261)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.006261, 0.007215)	                    	Count: 0 (0.000000%)	Total: 1 (0.005538%)
[0.007215, 0.008169)	                    	Count: 281 (1.556269%)	Total: 282 (1.561808%)
[0.008169, 0.009123)	                    	Count: 73 (0.404298%)	Total: 355 (1.966105%)
[0.009123, 0.010077)	 ###################	Count: 17197 (95.242577%)	Total: 17552 (97.208687%)
[0.010077, 0.011031)	                    	Count: 151 (0.836287%)	Total: 17703 (98.044968%)
[0.011031, 0.011985)	                    	Count: 266 (1.473194%)	Total: 17969 (99.518166%)
[0.011985, 0.012939)	                    	Count: 86 (0.476296%)	Total: 18055 (99.994461%)
[0.012939, 0.013893]	                    	Count: 1 (0.005538%)	Total: 18056 (100.000000%)
E0315 15:39:30.106456 10492 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101069 s, recommended is [0.0005, 0.05] s with no jitter.
I0315 15:39:30.106760 10492 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 1791  Min: 0.100548  Max: 0.101069  Mean: 0.100801
[0.100548, 0.100600)	                    	Count: 10 (0.558347%)	Total: 10 (0.558347%)
[0.100600, 0.100652)	                    	Count: 7 (0.390843%)	Total: 17 (0.949190%)
[0.100652, 0.100704)	                    	Count: 10 (0.558347%)	Total: 27 (1.507538%)
[0.100704, 0.100756)	                   #	Count: 51 (2.847571%)	Total: 78 (4.355109%)
[0.100756, 0.100809)	       #############	Count: 1190 (66.443329%)	Total: 1268 (70.798439%)
[0.100809, 0.100861)	               #####	Count: 478 (26.689001%)	Total: 1746 (97.487434%)
[0.100861, 0.100913)	                    	Count: 21 (1.172529%)	Total: 1767 (98.659966%)
[0.100913, 0.100965)	                    	Count: 10 (0.558347%)	Total: 1777 (99.218315%)
[0.100965, 0.101017)	                    	Count: 9 (0.502513%)	Total: 1786 (99.720825%)
[0.101017, 0.101069]	                    	Count: 5 (0.279174%)	Total: 1791 (100.000000%)

[image]
why does the track differ from the real movement?

Duke-Allen, 2019-03-22:
@athtest800 
I have solved it. Thanks for your help!

athtest800, 2019-03-24:
Nice! Is it possible to upload the final corrected .bag file?

Duke-Allen, 2019-03-25:
@athtest800 
There is the [bag](https://share.babel.cc/share.do?s=405003510720116). Because of the deviation of my IMU, so I have to set the angular velocity of x and y zero. I will try it again by a precise IMU. Thanks for your help again.