cartographer-project/cartographer_ros #1189: Scans flip while building the first submap

precyon, 2019-02-12:
I have managed to get `cartographer_ros` working for me reasonably well for the mapping bit except for a small glitch at the start. After beginning to create the first submap, the scans flip. The following GIF shows what I am talking about:

[image]

The mapping continues well after this flip and no issue is seen. But I can't seem to get my head around why this flip happens. 

1. Output of `rosbag_validate`: https://gist.github.com/precyon/2a5e0efffcae84d184028b78601535ff
2. Repo with all the configuration files:  https://github.com/precyon/cartographer_test 
    which is based on `cartographer_mir@6b228cd`
    
    Also, I use `cartographer@1.0.0` and `cartographer_ros@1.0.0`

3. The bagfile: https://drive.google.com/file/d/1WemcOkNY5FjCHBjdKSQ5MnP0kXWnBkqo/view?usp=sharing

I run,
`roslaunch cartographer_test 2d_mapper_demo.launch bag_filename:=test_take1.bag`

Thanks!

pvishal, 2019-02-13:
@precyon, I had a look at your files and noticed that you have rotated your IMU frame ( id: `vectornav`) by 180 degrees. Is it to account for the way your IMU is mounted? 

Like mentioned in #990, try breaking up your URDF into an having an intermediate `imu_link` which is at the same position as `vectornav` but rotated to be Z-up. Use `imu_link` as the tracking frame and see if this resolves your issue.

And since #990 is closed, see if your problem persists in the the latest versions.

precyon, 2019-02-13:
Thanks! I will try this and report back.

ojura, 2019-02-13:
I recommend just changing your  tracking frame to base_link. If your base_link is not z-up, make another child frame of base_link which is z-up and use that as the tracking frame.

closed 2020-10-27 by MichaelGrupp

sontung1010, 2024-04-21:
> @precyon, I had a look at your files and noticed that you have rotated your IMU frame ( id: `vectornav`) by 180 degrees. Is it to account for the way your IMU is mounted?
> 
> Like mentioned in #990, try breaking up your URDF into having an intermediate `imu_link` which is at the same position as `vectornav` but rotated to be Z-up. Use `imu_link` as the tracking frame and see if this resolves your issue.
> 
> And since #990 is closed, see if your problem persists in the latest versions.

Hey @pvishal which file did you see the 180 degree flip? I am having the same issue. Thanks