cartographer-project/cartographer_ros #1191: Correctly Configuring your Velodyne for use with Cartographer

jmount1992, 2019-02-14:
Hi,

Newbie to Cartographer here, but really hoping it can help solve my mapping problems. Currently having an issue with ensuring my velodyne pointcloud is in in the right format. I have experience with ROS and 2D LIDAR based SLAM solutions.

We recorded a small dataset from a Velodyne 32C using the default ROS configuration to ensure that it could work alongside Cartographer. However, when I run cartographer_rosbag_validate I get the following output:

```
E0214 16:09:42.326285 27848 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.123799 s, recommended is [0.0005, 0.05] s with no jitter.
I0214 16:09:42.326808 27848 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
Count: 495  Min: 0.116380  Max: 0.123799  Mean: 0.120103
[0.116380, 0.117122)	                    	Count: 1 (0.202020%)	Total: 1 (0.202020%)
[0.117122, 0.117864)	                    	Count: 0 (0.000000%)	Total: 1 (0.202020%)
[0.117864, 0.118606)	                    	Count: 0 (0.000000%)	Total: 1 (0.202020%)
[0.118606, 0.119348)	                    	Count: 0 (0.000000%)	Total: 1 (0.202020%)
[0.119348, 0.120090)	             #######	Count: 168 (33.939392%)	Total: 169 (34.141415%)
[0.120090, 0.120831)	       #############	Count: 325 (65.656563%)	Total: 494 (99.797981%)
[0.120831, 0.121573)	                    	Count: 0 (0.000000%)	Total: 494 (99.797981%)
[0.121573, 0.122315)	                    	Count: 0 (0.000000%)	Total: 494 (99.797981%)
[0.122315, 0.123057)	                    	Count: 0 (0.000000%)	Total: 494 (99.797981%)
[0.123057, 0.123799]	                    	Count: 1 (0.202020%)	Total: 495 (100.000000%)
```
I have not tried using this data to build a map as it the "fails" the rosbag validation. Is this recommendation valid? Can a map be built even when the data does not satisfy this recommendation?

I believe this "error" has something to do with the following statement, which can be found in the [documentation](https://google-cartographer-ros.readthedocs.io/en/latest/your_bag.html#validate-your-bag). Unfortunately, I am not 100% certain what this statement means.

> For example, with a Velodyne LIDAR, it is recommended to have one sensor_msgs/PointCloud2 message per UDP packet sent by the sensor instead of one message per revolution.

Additionally based on the comments in [here](https://github.com/googlecartographer/cartographer_ros/issues/904#issuecomment-401390088) it may have something to do with setting the npackets for the velodyne driver. However, I'm uncertain on the correct way to do this. And based on what I have seen in other issues this is fairly common occurrence. If anyone could shed some in-depth details/instructions that would be greatly appreciated, and may help many people using cartographer.

I would like for my small test data to pass this rosbag validation step before moving onto larger more complicated experiments. Any help is greatly appreciated.

Cheers

Semih67, 2019-02-19:
You have to record it again. And this error appears if you e.g. send messages through the network before you record. It is recommended to record it on the device that is connected to the sensors.

jmount1992, 2019-02-19:
@Semih67 thanks for getting back to me. I understand I have to re-record it. However, I forgot to mention my Velodyne is directly connected to the computer I am recording the data on. Any thoughts? Do you know anything about this npackets parameter?

ojura, 2019-02-19:
Hmm. The Velodyne ROS driver defaults to publishing one cloud per revolution. If you change it to emit one cloud per 'npackets', the good thing is that you have better timing, but then you have to estimate how many clouds there are per revolution, which is annoying. (In the example tuning for Google's bags, they set Cartographer to accumulate 160 clouds into a single cloud which is then scan matched).

I like having a single cloud per revolution, though. Maybe we could add computing per-point time based on the horizontal angle like loam: https://github.com/laboshinl/loam_velodyne/blob/master/src/lib/MultiScanRegistration.cpp#L205-L228

jmount1992, 2019-02-19:
Cheers @ojura for the insight. Should the ROS Velodyne driver work out of the box with Google Cartographer? There just seems to be a lot of chatter in various issues regarding Velodyne's, npackets and the delta between scans and the rosbag validation tool.

ojura, 2019-02-19:
As I mentioned, the default setting of the Velodyne ROS driver is one cloud per sweep. This means that you need to change the example 3D lua tuning file not to accumulate clouds (ie. change the number of accumulated clouds to be 1 instead of 160).

There is also support in cartographer_ros for an additional PointCloud2 channel called "time". You can fill it with the relative per-point time with respect to the cloud timestamp (the one in the ROS message header). It would be ideal if we could get the Velodyne driver to calculate this for us. We could do it in cartographer_ros based on the horizontal angle like loam does. (Not sure if the maintainers would approve :-))

jmount1992, 2019-02-27:
Thanks for the information @ojura. I did see that you need to change the lua configuration file. What I'm trying to get at is that when performing the rosbag validation it spits our the following as part of the message 

`recommended is [0.0005, 0.05] s`

This translates into a scan rate of somewhere between 2000 to 20Hz. Unfortunately I don't believe the Velodyne will be able to achieve 20Hz, so there must be something else going on. Because I doubt many LIDARs will be able to achieve scan rates higher than 20Hz.

Does the rosbag validation tool have some form of configuration file as well that it is basing the validation on?

ojura, 2019-02-27:
I have used Cartographer successfully with an 8 Hz lidar (2D though), so I don’t think you should take these warnings too literally. I think that Velodyne at 10 Hz is also ok.

jmount1992, 2019-02-28:
Cheers for the info mate. Might be nice if maybe there was a user updated table somewhere specifying the sensor combination that worked for them, including message rates, measurement accuracy (if possible), and the environment they tested it in.

Akash-Pardasani, 2019-10-19:
@jmount1992 Did you get it running? Can you share the final configuration that worked out for you?

jmount1992, 2019-10-19:
@Akash-Pardasani unfortunately due to time constraints we had to take an alternative, more simplified, approach. I do hope you get it working.

Akash-Pardasani, 2019-10-19:
@jmount1992 Thanks for the speedy reply. Do you mean an alternative, simplified approach to 3D mapping? I would actually love to know if there is one.

jmount1992, 2019-10-20:
@Akash-Pardasani no not an alternative to 3D mapping. Sorry that I cannot help you mate.

luanjinlu, 2020-03-12:
> As I mentioned, the default setting of the Velodyne ROS driver is one cloud per sweep. This means that you need to change the example 3D lua tuning file not to accumulate clouds (ie. change the number of accumulated clouds to be 1 instead of 160). Other than that, it should work out of box.
> 
> There is also support in cartographer_ros for an additional PointCloud2 channel called "time". You can fill it with the relative per-point time with respect to the cloud timestamp (the one in the ROS message header). It would be ideal if we could get the Velodyne driver to fill this in for us. We could also do it in cartographer_ros based on the horizontal angle like loam does. (Not sure if the maintainers would approve :-))

Thanks @ojura , this is right, I have validited with  Velodyne 64 LiDAR data, and cartographer can work well after set "TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1", this info has brought me great help.

closed 2020-10-23 by MichaelGrupp