cartographer-project/cartographer_ros #1192: demo error _ZN6google7LogSink8ToStringB5cxx11EiPKciPK2tmS2_m

oeyyey, 2019-02-14:
try to run Deutsches Museum demo (https://google-cartographer-ros.readthedocs.io/en/latest/demos.html)
```
wget -P ~/Downloads https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/cartographer_paper_deutsches_museum.bag
roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag
```
and then get this output
```
/opt/ros/kinetic/lib/cartographer_ros/cartographer_node: symbol lookup error: /opt/ros/kinetic/lib/cartographer_ros/cartographer_node: undefined symbol: _ZN6google7LogSink8ToStringB5cxx11EiPKciPK2tmS2_m

[cartographer_node-3] process has died [pid 11599, exit code 127, cmd /opt/ros/kinetic/lib/cartographer_ros/cartographer_node -configuration_directory /opt/ros/kinetic/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/oey/.ros/log/19c757f6-302e-11e9-94fb-d8c497604864/cartographer_node-3.log].
log file: /home/oey/.ros/log/19c757f6-302e-11e9-94fb-d8c497604864/cartographer_node-3*.log
```

BMannell, 2019-07-24:
I'm getting the same error. Any luck on resolving it?

oMalyugina, 2019-08-20:
I also got the error. Does someone have a solution?

BMannell, 2019-08-20:
I ended up doing a clean install of Ubuntu and it works now.

oeydev, 2019-08-21:
> I ended up doing a clean install of Ubuntu and it works now.

I aslo did that too. And upgrade to Ubuntu 18.04.

oMalyugina, 2019-08-23:
Yes, reinstallation Ubuntu 18 helped me! Thank you!

reinstallation ubuntu 16 saved errors and did nothing.

artemiialessandrini, 2020-07-07:
Any better solutions?:D

closed 2020-10-23 by MichaelGrupp