cartographer-project/cartographer_ros #1193: StartTrajectory service should support initial pose

gaschler, 2019-02-15:
The initial pose feature is the recommended way to use localization, in particular on larger maps.

Currently, `initial_pose` is only available through the `start_trajectory` executable.
More details are the in [PR where initial_pose was introduced](https://github.com/googlecartographer/cartographer_ros/issues/579#issuecomment-351482589).

ROS service `StartTrajectory.srv` should take initial pose as an argument (lua string should not be necessary).
This is a missing feature.
At the same time, the format of the initial pose argument could be reworked, it is unnecessarily complex to set the pose in 2D.

Ideally, the format should be so easy that start_trajectory_main.cc can be deleted.

neomanic, 2019-02-28:
When you say the format of the pose argument is overly complex, do you mean only for calling the executable rather than the `.srv` (which is loaded with parameters already)?

Would the best approach to adding `initial_pose` to `StartTrajectory.srv` be with a `geometry_msgs/Pose initial_pose` field? i.e. Okay to depend on the standard ros messages?

MichaelGrupp, 2019-02-28:
My 2 cents since we use this service in production since more than a year:

Initial pose would be easy to add but is not really the bottleneck IMO.

The ugliest part of this service is the binary string of the trajectory options proto, which is impossible to build by client code that should only know the message contract from `cartographer_ros_msgs`. The `start_trajectory_main.cc` executable makes this convenient because it hides exactly these implementation details away from the user/client with a proper interface that has only 3 simple options.

My ideal message definition would follow that interface:
```
geometry_msgs/Pose initial_pose
string configuration_directory
string configuration_basename
---
cartographer_ros_msgs/StatusResponse status
int32 trajectory_id
```
and all the configuration parsing, proto etc details are handled by the node. Then I'd be happy to see `start_trajectory_main.cc` die, but before: nah

gaschler, 2019-03-01:
My issue description was misleading.
The real issue is what @MichaelGrupp said, building the trajectory options proto is not user-friendly.
The need for start_trajectory_main.cc proves that.

MichaelGrupp, 2019-03-01:
I can do a PR for the refactored service as I wrote above, should be not too hard and would simplify things a lot.

MichaelGrupp, 2019-03-04:
Or, because this would be breaking existing API, a short RFC summarizing of what is written here.

neomanic, 2019-03-05:
I like your proposed solution @MichaelGrupp, that would suit our use case (lost localisation, need new trajectory with the same options as we started up with, just a new initial pose) well. In the meantime I'll just call out to the start_trajectory executable.

MichaelGrupp, 2019-03-05:
I suggest we continue the discussion here: https://github.com/googlecartographer/rfcs/pull/40

NTUwilliam, 2019-03-13:
@gaschler @MichaelGrupp @neomanic 
Firstly, thanks for you guys great work !

As @gaschler mentioned, detailes about the `initial_pose` is in the [PR where initial_pose was introduced](https://github.com/googlecartographer/cartographer_ros/issues/579#issuecomment-469160159). 
Although I've successfully set the initial **translation** for the `pure_localization` process, I still find it's very hard for the robot to localize itself in large symmetric environment without the **rotation** information(mainly the **`Yaw`** angle). 
And I'm still not clear about what determines the **orientation** of the `/map` frame, as shown in [issue /1218](https://github.com/googlecartographer/cartographer_ros/issues/1218), so it's quite hard for me to find the correct rotation information. 

So I'm wondering how can I get the **rotation** information (mainly the **`Yaw`** angle) to more accurately initialize the robot?

Any hint or comment is appreciated.
Thanks a lot !!

MichaelGrupp, 2020-10-23:
Implemented on master.

closed 2020-10-23 by MichaelGrupp