cartographer-project/cartographer_ros #1194: Apparent delay in map->robot TF

francescoricciuti, 2019-02-15:
Hi, 
I'm using cartographer 3d with a Robosense RS-16 on an Nvidia Jetson TX2.

While performing pure localization, I notice a strange behaviour of my Lidar informations. In the video here
[vokoscreen-2019-02-15_13-07-02.avi.zip](https://github.com/googlecartographer/cartographer_ros/files/2869181/vokoscreen-2019-02-15_13-07-02.avi.zip)
the behaviour: the green Pointcloud is the scan_matched_points from Cartographer, while the orange one is the raw one coming from my Lidar, transformed in the "map" frame through the TFs published by cartographer from my urdf file.

My first guess was that I was publishing my TFs too slowly, but I have the same behaviour also with all the TFs going at more than 100Hz.

Any guess of what could it be?

Thanks!

Semih67, 2019-02-19:
It looks like a problem in the global slamming. So I my first intention would be to increase the rotation weights in the POSE_GRAPH

francescoricciuti, 2019-02-20:
Thanks for the answer @Semih67 !

I tried following your suggestion, but I can see no changes neither increasing nor decreasing the rotation weight in POSE_GRAPH.optimization_problem.rotation_weight.

I don't think its a global slamming problem though. The scan_matched_points behave in fact as I would expect, and although sometimes they are not perfectly superimposed to the map while rotating , the behaviour is still good for my application.

What I wouldn't like to see is the raw Pointcloud rotating that much with the abrupt changes in yaw. That isn't the behaviour I would expect, as I have a TF from my Lidar frame to the map which is going at 50 Hz  (see my tf tree here attached [frames.pdf](https://github.com/googlecartographer/cartographer_ros/files/2883634/frames.pdf)).

I initially thought it was a visualization problem on Rviz, but that isn't the case because I have some logics relying on that transformations to be "good enough" and they fail when abruptly rotating.

It looks like if the "content" of the TF is slow in updating, and then even if the TF published by Cartographer goes at almost 150Hz, the result is a delay in the transformation. Is there somewhere I can see/check/set this behaviour?

Here my lua file:
```
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "imu_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_3D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_3D.submaps.num_range_data = 250
TRAJECTORY_BUILDER_3D.submaps.high_resolution = 0.20
TRAJECTORY_BUILDER_3D.submaps.high_resolution_max_range = 20
TRAJECTORY_BUILDER_3D.submaps.range_data_inserter.hit_probability = 0.55
TRAJECTORY_BUILDER_3D.submaps.range_data_inserter.miss_probability = 0.49

TRAJECTORY_BUILDER_3D.ceres_scan_matcher.translation_weight = 5e-3
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.rotation_weight = 4e-4

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 6
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 100
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

```

hvpandya-TRI, 2019-03-13:
To check the actual tf "lag" try the following:

` rosrun tf tf_monitor /map /odom`

if your stack does not have an odom frame, use the base_link instead

maxschnettler, 2019-07-24:
@francescoricciuti Had you been able to solve the issue? It sounds like a similar problem I am experiencing. Unfortunately, I am not able to play your video.

francescoricciuti, 2019-07-25:
@maxschnettler yes, I solved the problem.

It seems that I had some problems with my odometry, more in particular with the yaw, and that was causing a sort of delay in the rotation. I actually didn't understand if it was a problem of tf, actual data in the odom message, or a combination of both.

My solution was avoiding the use of odometry and retuning Cartographer to work without an odometry source, so that I avoid both dynamic TFs, working only with TFs published either by Cartographer or by the robot_state_publisher, and using data from the odom message. 

I will need to further investigate my odom, but this is not related to Cartographer so I will close the issue!

closed 2019-07-25 by francescoricciuti