cartographer-project/cartographer_ros #1196: roslaunch doesnt finish

Colidar, 2019-02-15:
hi all,

i'm using ubuntu 18.04 and ROS Melodic with an RPLIDAR Laserscan.
i have succesfully installed google cartographer and the demo of deutsches museum. 
Now i wanted to run cartographer with my own bag file which i uploaded on dropbox: 

**https://www.dropbox.com/sh/yutejspswzqqk8o/AACX8RzWfwWcmFipI0zXYRRQa?dl=0** 

with 

**~/catkin_ws$ roslaunch cartographer_ros my_robot.launch 2019-02-15-14-48-52.bag:=/home/corinna/bagfiles**

This is the following output:

... logging to /home/corinna/.ros/log/40dbd880-3128-11e9-a27a-b0359fce09c3/roslaunch-biws-nb05-3031.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://biws-nb05:39741/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [3046]
process[cartographer_node-2]: started with pid [3047]
process[cartographer_occupancy_grid_node-3]: started with pid [3048]
[ INFO] [1550238656.582974677]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_3d.lua' for 'backpack_3d.lua'.
[ INFO] [1550238656.583893625]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1550238656.583948525]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1550238656.584021609]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1550238656.584061032]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1550238656.584208312]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1550238656.584248917]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1550238656.584308388]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1550238656.584345697]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1550238656.584528757]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1550238656.584568942]: I0215 14:50:56.000000  3047 configuration_file_resolver.cc:41] Found '/home/corinna/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1550238656.590901911]: I0215 14:50:56.000000  3047 submap_3d.cc:321] Added submap 1
[ INFO] [1550238656.590987584]: I0215 14:50:56.000000  3047 map_builder_bridge.cc:130] Added trajectory with ID '0'.

i waited for 1h but nothing more happens but there is also no error or warning.
Can anyone help me to let it run to the end? Im only beginner level so please explain as easy as possible :+1:

Semih67, 2019-02-18:
I think the map is not published, thats why you do not get any output

Colidar, 2019-02-18:
> I think the map is not published, thats why you do not get any output

thanks semih67
how do i publish the map? Do i have to change a parameter im my_robot.lua:

-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 0.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 100

MAP_BUILDER.use_trajectory_builder_3d = false
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options

Semih67, 2019-02-19:
You say you are using your own bag file. So you have your own sensors? If yes which one? I guess it's just a laser scanner? And I see you are using the 3d cartographer mode ("TRAJECTORY_BUILDER_3D"). If that's the case you would also need a imu. Because in 3d imu is required. Otherwise I would recommend to work with 2d cartographer. And did you build your tf tree?

Colidar, 2019-02-21:
you were right, i copied the wrong file for 2d cartographer. Thanks!! 
Yes i only have one rplidar a2 laserscanner (https://www.slamtec.com/en/Lidar/A2) without imu. And i only want to use the lidar on my table without any robot yet. Thats why i thought i dont need a tf tree. Or do i need one? Can i easily create a tf tree for only laserscan? The tutorials on ros wiki look quiet complicated.

joyde3382, 2019-03-14:
Having exactly the same issue, did you manage to find a solution?

MrGitGo, 2019-03-26:
Your urdf file creates a tf-tree for you. And yes the tf tree is alwo required. In the urdf you have to write with which sensors you are working, which components your robot has...

joyde3382, 2019-03-28:
@MrGitGo I see what you mean, however i'm not quite sure how to configure the files. Right now my "robot" is purely an RPLidar and nothing else.

This is my current rqt_graph.

[image]

Also would help alot if i could take a look at your .lua .launch and .urdf files :)

Colidar, 2019-04-04:
hey @joyde3382, 
now i managed to run cartographer (with a boat example) in real time, see my files on dropbox:
https://www.dropbox.com/sh/yutejspswzqqk8o/AACX8RzWfwWcmFipI0zXYRRQa?dl=0

to run real time cartographer:
1. Terminal: roslaunch rplidar_ros rplidar.lauch
2. Terminal: roslaunch cartographer_ros demo_boat.launch 

But my tf is still missing at my current rqt graph, i don't know how to add it. Maybe you know @joyde3382 because your rqt graph looks complete!

joyde3382, 2019-04-08:
@Colidar Awesome, got it to work! Thanks for your help. Do you by any chance have experience in using the generated cartographer map for robot navigation? I'd like to make a robot that explores an area autonomously based on the cartographer map.

Colidar, 2019-04-08:
@joyde3382 great!
I'd like to do the same but I don't have any experience. But i would be happy if we help each other!
My next step is to integrate IMU

closed 2019-06-25 by Colidar