cartographer-project/cartographer_ros #1199: Running 3D Cartographer with IMU and 3D LiDAR

Duke-Allen, 2019-02-20:
Hello, I'm running cartographer to build 3D map by using IMU and a 16-lines lidar. When I running cartographer by my data bag, there is a error:

> setting /run_id to ee3da2f8-3513-11e9-8749-5e7f548b2e15
process[rosout-1]: started with pid [14834]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [14838]
process[cartographer_node-3]: started with pid [14851]
process[cartographer_occupancy_grid_node-4]: started with pid [14856]
process[rviz-5]: started with pid [14860]
**[ERROR] [1550669534.733878883]: Error reading end tag.**
[ INFO] [1550669534.772017127]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer_ros/configuration_files/rslidar_scan_3d.lua' for 'rslidar_scan_3d.lua'.
[ INFO] [1550669534.772830641]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1550669534.772952050]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1550669534.773149716]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1550669534.773357015]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1550669534.773729083]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1550669534.773822748]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1550669534.773989054]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1550669534.774085055]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1550669534.774347651]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1550669534.774438660]: I0220 21:32:14.000000 14851 configuration_file_resolver.cc:41] Found '/home/nvidia/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1550669534.798182173]: I0220 21:32:14.000000 14851 submap_3d.cc:321] Added submap 1
[ INFO] [1550669534.798454368]: I0220 21:32:14.000000 14851 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[robot_state_publisher-2] process has died [pid 14838, exit code 255, cmd /opt/ros/kinetic/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/nvidia/.ros/log/ee3da2f8-3513-11e9-8749-5e7f548b2e15/robot_state_publisher-2.log].
log file: /home/nvidia/.ros/log/ee3da2f8-3513-11e9-8749-5e7f548b2e15/robot_state_publisher-2*.log
=================

my configuration files:
[demo_rslidar_3d.launch:](https://github.com/Duke-Allen/cartographer-configuration.git)
launch
  param name="/use_sim_time" value="true" 
 include file="$(find cartographer_ros)/launch/rslidar_3d.launch" 
 node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/mydemo_3d.rviz"  
/launch

[rslidar_3d.launch:](https://github.com/Duke-Allen/cartographer-configuration.git)
launch 
  param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/rslidar_2d.urdf" 

  node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" 

  node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename rslidar_scan_3d.lua"
      output="screen"
      <remap from="/points2" to="/rslidar_points" />

  /node

  node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" 
/launch

[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration.git)

[rslidar_scan_3d.lua:](https://github.com/Duke-Allen/cartographer-configuration.git)
`include "map_builder.lua"
include "trajectory_builder.lua"
options = {
 map_builder = MAP_BUILDER,
 trajectory_builder = TRAJECTORY_BUILDER,
 map_frame = "map",
 tracking_frame = "base_link",
 published_frame = "base_link",
 odom_frame = "odom",
[... 186 lines cut ...]
  <joint name="horizontal_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="rslidar" />
    <origin xyz="0 -0.075 0.04" rpy="0 0 0" />
  </joint>

  <link name="imu">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <cylinder length="0.05" radius="0.01" />
      </geometry>
      <material name="yellow" />
    </visual>
  </link>
  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu" />
    <origin xyz="0 0 0.04" rpy="0 0 0" />
  </joint>
</robot>`

[assets_writer_rslidar_3d.lua](https://github.com/Duke-Allen/cartographer-configuration.git)

Thank you for your help.

Duke-Allen, 2019-02-21:
I found the error in rslidar_2d.urdf file, and correct it. But it not build map.
This is my data bag: 
[image]
running the cartographer:
[image]

[image]

I don't know how to solve, thanks for your help.

Duke-Allen, 2019-02-22:
@cschuet @damonkohler @ojura @SirVer @gaschler 
I found the solution by setting imu_sampling_ratio less than 1. But it died when the bag plays toward the end:
> [ INFO] [1550838613.792574978, 1550636672.551481728]: I0222 20:30:13.000000  8565 pose_graph_3d.cc:610] Rotational residuals histogram:
Count: 6236  Min: 0.006697  Max: 3.104059  Mean: 0.608558
[0.006697, 0.316433)	               #####	Count: 1492 (23.925592%)	Total: 1492 (23.925592%)
[0.316433, 0.626169)	            ########	Count: 2458 (39.416294%)	Total: 3950 (63.341885%)
[0.626169, 0.935906)	                ####	Count: 1144 (18.345093%)	Total: 5094 (81.686981%)
[0.935906, 1.245642)	                  ##	Count: 575 (9.220654%)	Total: 5669 (90.907631%)
[1.245642, 1.555378)	                   #	Count: 357 (5.724823%)	Total: 6026 (96.632454%)
[1.555378, 1.865114)	                    	Count: 146 (2.341244%)	Total: 6172 (98.973701%)
[1.865114, 2.174851)	                    	Count: 10 (0.160359%)	Total: 6182 (99.134064%)
[2.174851, 2.484587)	                    	Count: 16 (0.256575%)	Total: 6198 (99.390633%)
[2.484587, 2.794323)	                    	Count: 19 (0.304682%)	Total: 6217 (99.695320%)
[2.794323, 3.104059]	                    	Count: 19 (0.304682%)	Total: 6236 (100.000000%)
[ INFO] [1550838614.752949334, 1550636673.510676095]: I0222 20:30:14.000000  8565 pose_graph_3d.cc:397] Remaining work items in queue: 2403
[ INFO] [1550838614.756068932, 1550636673.510676095]: I0222 20:30:14.000000  8467 collated_trajectory_builder.cc:72] imu rate: 37.14 Hz 2.69e-02 s +/- 1.14e-02 s (pulsed at 89.54% real time)
[ INFO] [1550838614.756212150, 1550636673.510676095]: I0222 20:30:14.000000  8467 collated_trajectory_builder.cc:72] points2 rate: 9.92 Hz 1.01e-01 s +/- 1.97e-05 s (pulsed at 97.52% real time)
[cartographer_node-3] process has died [pid 8467, exit code -9, cmd /home/nvidia/carto_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/nvidia/carto_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename rslidar_scan_3d.lua /points2:=/rslidar_points __name:=cartographer_node __log:=/home/nvidia/.ros/log/af0ebfb0-369c-11e9-9cb7-3e1dbfe54845/cartographer_node-3.log].
log file: /home/nvidia/.ros/log/af0ebfb0-369c-11e9-9cb7-3e1dbfe54845/cartographer_node-3*.log

[image]
[image]
I don't know why there is a error of pointcloud status. Could you help me?

closed 2019-02-26 by Duke-Allen

christophTUM, 2021-06-21:
How did you solve this?