cartographer-project/cartographer_ros #1200: struggling with running Cartographer on my own .bag

goeldisandro, 2019-02-20:
I m relatively new to ROS and thank you in advance for your help

just as a reference; as soon as cartographer is working with the point cloud data i d like also to include the odom data an do a few studies about the quality of the generated Maps.

I went through the tutorials from Google Cartographer  (https://google-cartographer-ros.readthedocs.io/en/latest/demos.html). As long as i tried it with the given .bag files everything went well.

Now i am trying to launch it with my own .bag file, which is just a short example and only recorded with a velodyne VLP 16. The important files are located in; https://github.com/goeldisandro/SLAM

The validation of the .bag did not showed some concerns as far as i see.

When i now try to launch it with the command;
roslaunch cartographer_ros offline_my_robot.launch bag_filenames:=${HOME}/bagfiles/2019-01-10-14-37-16.bag
rviz shows up but with the Global Status: Warn
No tf data.  Actual error: Fixed Frame [map] does not exist

I think that the problem might occur, because i do not have a tf transformation.
I already made the tf tutorials from ROS, but i do not know how to adapt it on my Situation.
Is there a place, where i find some similar projects? Or do i just have to make a change in RViz?

everything is launched on a linux laptop with ros kinetic

Thanks for your help and greetings from Switzerland

Sandro

jmount1992, 2019-02-27:
Hi Sandro (@goeldisandro),

I am a newbie to Cartographer as well, but I might be able to help with the transform part. In ROS are you publishing somewhere the transforms between your robot's coordinate frame (typically called base_link) and the sensors' coordinate frames (typically called <sensor>_link, e.g. imu_link or scan_link)? 

There is a bit of information about the transforms (or TFs) required by Cartographer [here](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#required-tf-transforms) .  Based on my understanding of that info, you will need to at least provide the robot to sensor transforms.

You might be able to help me with my issue. What rate were you running your VLP-16 at? I'm using a Velodyne 32C but get errors when running the validation tool, mainly that the delta between scans is too large. Did you have to change anything in the velodyne ros driver?

Cheers,
James

Colidar, 2019-03-07:
Hi @goeldisandro !
I have the same problem. Did you already solved it? 
I think we have to change the used topics in the transform part.
When running  <rosbag info cartographer_paper_deutsches_museum.bag> it says:

topics:  horizontal_laser_2d    : sensor_msgs/MultiEchoLaserScan
             imu                            : sensor_msgs/IMU
             vertikal_laser_2d        : sensor_msgs/MultiEchoLaserScan

And when I'm running <rosbag info mybagfile> is says:

topics: /rosout                       : rosgraph_msgs/Log
           /rosout_agg                : rosgraph_msgs/Log
           /scan                           : sensor_msgs/LaserScan

And when you run: <rosrun tf> while using cartographer you cant print the running transformations. View the created PDF with <evince frames.pdf>

But i don't know what and how to change...

goeldisandro, 2019-03-08:
Hi yes I was able to launch cartographer on my own .bag and everything went well.
The change, that i had to make were;
-change from tf to an urdf File, which you can find on my repositories
-made up an new workspace and used catkin build in stead of catkin_make

@Colidar i'm not sure if i understood your concerns completely....
Is the problem, that the needed topics from above arent published anymore, when you run the cartographer node?

@jmount1992 until now i only used the default Settings and made no changes, but it might be, that i need to adapt the setting soon. In case i will surely keep you up to date

Colidar, 2019-03-18:
@goeldisandro 
sorry for my unclear answer.. the commands i wrote down suddenly disappeared.. 
i used **rosbag info** to compare the published topics from demo bagfile (deutsches museum) and my own bagfile. I'm only using LaserScan (with an RPLidar A2) and not MultiEchoLaserscan (as deutsches museum), so i changed my_robot.lua, demo_my_robot.launch and the urdf file:

**LUA FILE:**
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "laser_link",
  published_frame = "laser_link",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 10,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 100
return options

**LAUNCH FILE:**
<launch>
  <param name="/use_sim_time" value="true" />

  <include file="$(find cartographer_ros)/launch/my_robot.launch" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />

</launch>

**URDF FILE:**:
<robot name="cartographer_backpack_2d">
  <material name="orange">
    <color rgba="1.0 0.5 0.2 1" />
  </material>
  <material name="gray">
    <color rgba="0.2 0.2 0.2 1" />
  </material>

  <link name="laser_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <cylinder length="0.05" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>
 
  <link name="base_link" />

  <joint name="scan_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="laser_link" />
    <origin xyz="0.1007 0 0.0558" />
  </joint>

</robot>

It's still not working..maybe you can find my mistake..

Here is the link to my bagfile:
https://www.dropbox.com/sh/yutejspswzqqk8o/AACX8RzWfwWcmFipI0zXYRRQa?dl=0

closed 2020-10-23 by MichaelGrupp

VigneshS1993, 2020-12-27:
Hello everyone!!
When I run cartoggrapher validate command for my rosbag I get some errors. Any idea where I would get some documentation on the error codes and how to solve them ?

VTOLIntern4, 2022-01-11:
> Hello everyone!! When I run cartoggrapher validate command for my rosbag I get some errors. Any idea where I would get some documentation on the error codes and how to solve them ?

Kindly put SS or copy the errors you're facing