cartographer-project/cartographer_ros #1201: Unstable trajectory rotation jumps during 3D pose graph optimization

jccurtis, 2019-02-20:
Our team is using 2 [VLP-16](https://velodynelidar.com/vlp-16.html)'s and an [Advanced Navigation Spatial](https://www.advancednavigation.com/product/spatial) as part of a research vehicle for 3D mapping. We have been using cartographer to perform 3D SLAM in a ROS environment to process the data streams in real-time and from rosbags. After spending some time following the tuning guide we have been unable to get global SLAM to produce stable results in real-time. 

I have found that the pose graph optimization (`POSE_GRAPH.optimize_every_n_nodes`) causes large (10's of degree) rotation jumps in our trajectory but these are eventually smoothed after 10-20 seconds (realtime) such that the final result is smooth and loop closed. If I disable the pose graph optimization, the local SLAM result is relatively smooth and stable but eventually drifts slightly over time (as expected without loop closure). This issue is particularly confusing because the gravity acceleration (visualized with [rviz](http://wiki.ros.org/rviz_imu_plugin)) doesn't align with the positive z axis during these rotational shifts even while driving on a smooth/flat road.

I've attached the configuration following the template and would like to see if there is some configuration parameter we have overlooked without needing to upload a bag:

1. `rosbag_validate` returns no `stdout` except a seg fault after processing a bag file. I know the tool is (mostly?) working because it does return output ([gist](https://gist.github.com/jccurtis/d7e41bbc3c8d56c4aa96db90f19abc9d#file-cartographer_rosbag_validate_output_2019-01-11-txt) if laser scans from the each lidar sensor are included in the bag along with the pointcloud the Velodyne driver also produces. I believe the validation tool can't tell the difference between these two messages sent from the same sensor because they share the same `tf`.
2. [lua configuration](https://gist.github.com/jccurtis/d7e41bbc3c8d56c4aa96db90f19abc9d#file-live_slam_3d_navsat-lua), [launch](https://gist.github.com/jccurtis/d7e41bbc3c8d56c4aa96db90f19abc9d#file-playback-launch), and [URDF](https://gist.github.com/jccurtis/d7e41bbc3c8d56c4aa96db90f19abc9d#file-robot-urdf).
3. See above note...

Thanks for any help or advice you can provide!

jccurtis, 2019-03-19:
Hi cartographer team. I'm just bumping this issue to see if you can provide any guidance on reducing discontinuous jumps resulting from the global SLAM optimization. Thank you!

gloomysunday, 2019-07-04:
Hi, @jccurtis. There are some constraints based on IMU observations of angular velocities and linear acceleration in the global SLAM optimization. The global optimization will optimize the parameters (node pose, submap pose, imu_calibration and gravity_constant) to minimize the cost function. When you start from a stationary pose, the delta velocity is too small, so that the parameters are overfit. 

You could find the imu cost optimization in cartographer/mapping/internal/optimization/optimization_problem_3d.cc : OptimizationProblem3D::Solve, 
problem.AddResidualBlock(AccelerationCostFunction3D::CreateAutoDiffCostFunction)
problem.AddResidualBlock(RotationCostFunction3D::CreateAutoDiffCostFunction)

I think there are several methods to solve this problem.
1. you could calculate the delta velocity to determine whether adding the imu cost to global optimization, if  delta velocity > threshold, adding imu cost;
2. The node pose is optimized to changing too much, so perhaps you could add a cost function based on  delta node pose to global optimization;
3. you could just ignore the problem, after all the trajectory is eventually smoothed.

MichaelGrupp, 2020-10-23:
> I have found that the pose graph optimization (POSE_GRAPH.optimize_every_n_nodes) causes large (10's of degree) rotation jumps in our trajectory but these are eventually smoothed after 10-20 seconds [...]

I do't see a problem here, this is exactly how a SLAM system should behave: smooth continuous odometry (local SLAM) and drift correction by loop closure (global SLAM).

closed 2020-10-23 by MichaelGrupp

wolfgangschwab, 2021-09-10:
@jccurtis , I just noticed that I'm facing a similar or the same issue (#1666) as you described here. 
Did you find a solution for your problem?
Did you also get a NO_CONVERGENCE notice in the log-file?