cartographer-project/cartographer_ros #1203: cartographer_pbstream_to_ros_map: Check failed: cairo_image_surface_get_format(surface.get()) == kCairoFormat (-1 vs. 0)

kajep09, 2019-02-21:
Hello

I have a problem with the cartographer_pbstream_to_ros_map script where sometimes it failes to generate a pgm/yaml file and fails with the error below:
```
rosrun cartographer_ros cartographer_pbstream_to_ros_map -pbstream_filename=/home/kaspercj/.ros/stoptest.pbstream -map_filestem=lalala
I0221 12:24:01.590298 21498 pbstream_to_ros_map_main.cc:43] Loading submap slices from serialized data.
I0221 12:24:01.599465 21498 pbstream_to_ros_map_main.cc:51] Generating combined map image from submap slices.
F0221 12:24:01.599550 21498 image.cc:55] Check failed: cairo_image_surface_get_format(surface.get()) == kCairoFormat (-1 vs. 0) 
*** Check failure stack trace: ***
    @     0x7f482d7655cd  google::LogMessage::Fail()
    @     0x7f482d767433  google::LogMessage::SendToLog()
    @     0x7f482d76515b  google::LogMessage::Flush()
    @     0x7f482d767e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x4de6a7  cartographer::io::Image::Image()
    @           0x4cf31d  cartographer_ros::(anonymous namespace)::Run()
    @           0x4cd8e8  main
    @     0x7f482bf72830  __libc_start_main
    @           0x4cf049  _start
    @              (nil)  (unknown)
Aborted (core dumped)

```

I have 2 different pbstream files I have recorded and (from what I remember ) a quick visual inspection told me that the map looked okay. However, since I see this error now and again,  I just pulled 2 files that crashes and works respectively.

https://drive.google.com/file/d/1Dj2pM2_rj1RXT6Lx6-r0QqO5MfOUiM3C/view?usp=sharing (not working)
https://drive.google.com/file/d/12Yw8IXI6WDsgynONO-GJnpNF7KrZKhGz/view?usp=sharing (working)

I have a suspicion that it has something to do with the size of the map/trajectory because it seems to be related to the filesize of the pbstream files (it often crashes with small files it seems). 

Can anyone shed some light on why the `   CHECK_EQ(cairo_image_surface_get_format(surface.get()), kCairoFormat);` check fails?

kajep09, 2019-03-14:
I think that I've found the reason for the crash, but I would like to know if my reasoning makes sense before closing this issue. 
Basically, what I think happens is that you cannot convert a pbstream file where the trajectory does not contain a single complete submap into a ros map.

When tuning my robot with a single laser scanner with an output of 30Hz I used a config with submaps.num_range_data = 200. If I drive around and never let the first submap get to 400 points (2*num_range_data) and then save the pbstream file and try to convert it, it always fails. However, if I let one submap complete then I haven't seen the issue with converting.

@gaschler @damonkohler @cschuet 
Can anyone confirm that this makes sense? If so (and if it's possible to do this) maybe there could be a warning or something in the `cartographer_pbstream_to_ros_map` script that tells the user what might be wrong.

gaschler, 2019-03-14:
@kajep09 Your explanation makes sense. 
The check could be if the submap list or image size is empty and then log fatal.
I'd be happy to review a pull request for this.

closed 2020-10-23 by MichaelGrupp

yyggnb, 2023-10-31:
@kajep09  I encountered the same problem as you ，how can i solve it? Just make parameters small or build a larger maps? Should i adjust parameters ?