cartographer-project/cartographer_ros #1208: Build cartographer from source (master)

ashnarayan13, 2019-02-26:
Hi, 

I am able to run cartographer when everything is in the 1.0.0 version for both cartographer and for cartographer_ros. 

But, I am unable to build it in master. Both cartographer and cartographer_ros give me abseil related errors. Although cartographer builds abseil itself. Is there like a guide as to how to build cartographer and cartographer_ros in master? 

`FAILED: : && /usr/bin/c++   -std=c++11 -O3 -DNDEBUG   CMakeFiles/cartographer_pbstream.dir/cartographer/io/pbstream_main.cc.o  -o cartographer_pbstream  -rdynamic libcartographer.a /home/priyamash/tufast/tufast_slam/install_isolated/lib/libceres.a /usr/local/lib/libglog.a /usr/local/lib/libgflags.a -lspqr -ltbbmalloc -ltbb -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt /usr/local/lib/libmetis.so -lcxsparse -llapack -lf77blas -latlas -lsuitesparseconfig -lrt /usr/local/lib/libmetis.so -lcxsparse -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog /usr/local/lib/libgflags.a -lpthread -lcairo /usr/local/lib/libprotobuf.a abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a abseil/src/abseil-build/absl/debugging/libabsl_symbolize.a abseil/src/abseil-build/absl/time/libabsl_time.a abseil/src/abseil-build/absl/strings/libstr_format_internal.a abseil/src/abseil-build/absl/strings/libstr_format_extension_internal.a abseil/src/abseil-build/absl/strings/libabsl_str_format.a abseil/src/abseil-build/absl/strings/libabsl_strings.a abseil/src/abseil-build/absl/hash/libabsl_hash.a abseil/src/abseil-build/absl/algorithm/libabsl_algorithm.a abseil/src/abseil-build/absl/base/libabsl_base.a abseil/src/abseil-build/absl/base/libabsl_dynamic_annotations.a abseil/src/abseil-build/absl/base/libabsl_internal_malloc_internal.a abseil/src/abseil-build/absl/base/libabsl_internal_spinlock_wait.a abseil/src/abseil-build/absl/base/libabsl_internal_throw_delegate.a abseil/src/abseil-build/absl/container/libabsl_container.a abseil/src/abseil-build/absl/container/libtest_instance_tracker_lib.a abseil/src/abseil-build/absl/debugging/libabsl_debugging.a abseil/src/abseil-build/absl/debugging/libabsl_examine_stack.a abseil/src/abseil-build/absl/debugging/libabsl_failure_signal_handler.a abseil/src/abseil-build/absl/debugging/libabsl_leak_check.a abseil/src/abseil-build/absl/debugging/libabsl_stack_consumption.a abseil/src/abseil-build/absl/debugging/libabsl_stacktrace.a abseil/src/abseil-build/absl/memory/libabsl_memory.a abseil/src/abseil-build/absl/meta/libabsl_meta.a abseil/src/abseil-build/absl/numeric/libabsl_int128.a abseil/src/abseil-build/absl/numeric/libabsl_numeric.a abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a abseil/src/abseil-build/absl/types/libabsl_any.a abseil/src/abseil-build/absl/types/libabsl_bad_any_cast.a abseil/src/abseil-build/absl/types/libabsl_bad_optional_access.a abseil/src/abseil-build/absl/types/libabsl_optional.a abseil/src/abseil-build/absl/types/libabsl_span.a abseil/src/abseil-build/absl/types/libabsl_variant.a abseil/src/abseil-build/absl/utility/libabsl_utility.a -lpthread -Wl,-rpath,/usr/local/lib: && :
libcartographer.a(serialization_format_migration.cc.o): In function `cartographer::io::MigrateStreamFormatToVersion1(cartographer::io::ProtoStreamReaderInterface*, cartographer::io::ProtoStreamWriterInterface*, bool)':
serialization_format_migration.cc:(.text+0x427e): undefined reference to `absl::container_internal::UnsampleSlow(absl::container_internal::HashtablezInfo*)'
libcartographer.a(serialization_format_migration.cc.o): In function `absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<int, std::vector<cartographer::mapping::proto::SerializedData, std::allocator<cartographer::mapping::proto::SerializedData> > >, absl::hash_internal::Hash<int>, std::equal_to<int>, std::allocator<std::pair<int const, std::vector<cartographer::mapping::proto::SerializedData, std::allocator<cartographer::mapping::proto::SerializedData> > > > >::~raw_hash_set()':
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEED2Ev[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEED5Ev]+0xcf): undefined reference to `absl::container_internal::UnsampleSlow(absl::container_internal::HashtablezInfo*)'
libcartographer.a(serialization_format_migration.cc.o): In function `absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<int, std::vector<cartographer::mapping::proto::SerializedData, std::allocator<cartographer::mapping::proto::SerializedData> > >, absl::hash_internal::Hash<int>, std::equal_to<int>, std::allocator<std::pair<int const, std::vector<cartographer::mapping::proto::SerializedData, std::allocator<cartographer::mapping::proto::SerializedData> > > > >::prepare_insert(unsigned long)':
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x370): undefined reference to `absl::container_internal::RecordInsertSlow(absl::container_internal::HashtablezInfo*, unsigned long, unsigned long)'
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x451): undefined reference to `absl::container_internal::global_next_sample'
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x4d3): undefined reference to `absl::container_internal::global_next_sample'
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x51a): undefined reference to `absl::container_internal::UnsampleSlow(absl::container_internal::HashtablezInfo*)'
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x524): undefined reference to `absl::container_internal::SampleSlow(long*)'
serialization_format_migration.cc:(.text._ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm[_ZN4absl18container_internal12raw_hash_setINS0_17FlatHashMapPolicyIiSt6vectorIN12cartographer7mapping5proto14SerializedDataESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x531): undefined reference to `absl::container_internal::UnsampleSlow(absl::container_internal::HashtablezInf [... cut ...] aESaIS7_EEEENS_13hash_internal4HashIiEESt8equal_toIiESaISt4pairIKiS9_EEE14prepare_insertEm]+0x538): undefined reference to `absl::container_internal::SampleSlow(long*)'
collect2: error: ld returned 1 exit status
[262/461] Building CXX object CMakeFiles/cartographer_test_library.dir/cartographer/mapping/internal/testing/test_helpers.cc.o
ninja: build stopped: subcommand failed.`

Is there a specific abseil version that cartographer requires? Are there any other dependencies that I miss? 

Thanks!

ojura, 2019-02-26:
Is this perhaps https://github.com/googlecartographer/cartographer_ros/issues/1075?

ashnarayan13, 2019-02-27:
Hi @ojura , yes this resolved the problem. Thank you! is it possible to have this somewhere in the documentation?

ojura, 2019-02-27:
Yes, that would indeed be very helpful. Any chance you could open a pull request for adding this to documentation?

MichaelGrupp, 2020-10-23:
I will try to fix the build documentation, see #1531

closed 2020-10-23 by MichaelGrupp