cartographer-project/cartographer_ros #1212: call rosservice start_trajectory in python results in invalid trajectory options

evenmarbles, 2019-02-28:
Hi,

I am trying to start a trajectory from python using the start_trajectory service. However, I am receiving the following error:

```
code: 3
message: "Invalid trajectory options."
```

I am calling the service as follows:
```
trajectory_options_msg = TrajectoryOptions(
            tracking_frame="imu_link",
            published_frame="odom",
            odom_frame="odom",
            use_odometry=True,
            num_laser_scans=1,
            num_subdivisions_per_laser_scan=1,
            rangefinder_sampling_ratio=1.,
            odometry_sampling_ratio=1.,
            fixed_frame_pose_sampling_ratio=1.,
            imu_sampling_ratio=1.,
            landmarks_sampling_ratio=1.)

        sensor_topics_msg = SensorTopics()
        sensor_topics_msg.laser_scan_topic = "/scan"
        sensor_topics_msg.imu_topic = "/imu/data"
        sensor_topics_msg.odometry_topic = "/odom"

        rospy.wait_for_service("/start_trajectory")
        try:
            self.trajectory_id = self.start_trajectory_proxy(trajectory_options_msg, sensor_topics_msg)
        except rospy.ServiceException as e:
            print("/start_trajectory service call failed: {}".format(e))
```

Considering [issue 729](https://github.com/googlecartographer/cartographer_ros/issues/729) I probably have to set `trajectory_builder_options_proto`. 

But how do I go about doing that in Python?

Any help would be appreciated! Thanks.

MichaelGrupp, 2019-02-28:
Use `subprocess` and call the [start trajectory executable](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/start_trajectory_main.cc) with 

```
rosrun cartographer_ros cartographer_start_trajectory --configuration_directory <your Lua config dir> --configuration_basename <your Lua config name>
```

This isn't the most pythonic way but it saves you from doing the proto stuff.

MichaelGrupp, 2020-10-19:
The whole /start_trajectory service was also refactored now (see current version on master) and doesn't require a proto string anymore.

closed 2020-10-19 by MichaelGrupp