cartographer-project/cartographer_ros #1214: cartographer_ros logging is messy

neomanic, 2019-02-28:
Now that we've got cartographer_ros up and running nicely, I was trying to tidy up the logging, since there's a whole ton of information that isn't necessary unless there are issues.

For reference, adding the following to your lua config reduces the bulk of it:
```
    -- turn off logging
    POSE_GRAPH.log_residual_histograms = false
    POSE_GRAPH.constraint_builder.log_matches = false
    POSE_GRAPH.optimization_problem.log_solver_summary = false
```

but I am still left with the following, there is no option to turn off the motion_filter and odom/scan rate reporting.

```
[ INFO] [1551394545.696911241]: I0301 09:55:45.000000 22677 motion_filter.cc:42] Motion filter reduced the number of nodes to 1.6%.
[ INFO] [1551394552.114027913]: I0301 09:55:52.000000 22677 collated_trajectory_builder.cc:72] odom rate: 50.00 Hz 2.00e-02 s +/- 8.25e-04 s (pulsed at 100.00% real time)
[ INFO] [1551394552.114580119]: I0301 09:55:52.000000 22677 collated_trajectory_builder.cc:72] scan rate: 124.51 Hz 8.03e-03 s +/- 5.06e-04 s (pulsed at 99.68% real time)
[ INFO] [1551394553.652137774]: I0301 09:55:53.000000 22677 submap_2d.cc:187] Added submap 2
```

But also, this wouldn't be an issue if cartographer_ros obeyed the standard ros `output:=screen` or `output:=log` commands. Currently all log messages go to standard out, so when we are running in production mode, they are found in both `cartographer-cartographer-40-stdout.log` *and*  `rosout.log`.

I haven't seen this addressed in any other issues.

MichaelGrupp, 2019-03-01:
You can raise the logger level of glog by appending this command line option:
```bash
--minloglevel 1  # 0 (INFO), 1 (WARNING), 2 (ERROR), 3 (FATAL)
```

or when the node is running:
```
rosservice call /cartographer/set_logger_level {"logger: 'ros', level: 'warn'"}
```

MichaelGrupp, 2019-03-01:
One more note: the first one overrules the second - so if you want to change the level during runtime the ROS way is the way to go.

neomanic, 2019-03-05:
Thanks @MichaelGrupp, that should solve my issue nicely. 

For clarity's sake, do you have any comments on all logging going to stdout as per my final remarks above? I'm guessing because it's using GLog, there's no say in the matter?

MichaelGrupp, 2019-03-05:
Don't know, maybe you can have a look at the ROS log sink which redirects glog to ROS logging.
https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/ros_log_sink.h

Duke-Allen, 2019-03-05:
@neomanic Hello neomanic, I am running 3D cartographer by a 16-lines lidar and IMU. I encounter some problems, so I would like to see how you get the correct result. 
My map is messy and my robotmodel is reversed frequently. It dose not correspond with the real environment. I have checked the urdf file, but I did not find the problem(or maybe I did not check it out). 

[my_lua file](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[urdf_file](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
launch file: [demo_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)    [rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
This is my real position about lidar and IMU:
[image]
[image]

Do you know the problem and how to solve it?

neomanic, 2019-03-08:
@Duke-Allen you should start a separate issue to discuss. 

I suspect you need to get your lidar and IMU working and viewable in Rviz without Cartographer first. Once those are set up correctly, then getting Cartographer running is pretty straightforward.

closed 2020-10-17 by MichaelGrupp