cartographer-project/cartographer_ros #1218: initial orientation of the map frame

NTUwilliam, 2019-03-08:
Hi all!

I've been using Cartographer for several months and have successfully accomplished `2D SLAM `and `2D pure localization` function. 
Now I'm trying to build a map and later set the `initial_pose` for `pure_localization`. But when I started, the **orientation** of the `/map` frame is not correct. 
The correct **orientation** of the `/map` frame should be the same with the **orientation** of the `/base_link` frame (as shown below, the orientation deviation/error between `/map` frame and `/base_link` frame is around 60 degrees):
[image]

And the **`/tf`** between **`/map`** frame and **`/base_link`** frame is keep changing like this:

```
williamwoo@williamwoo:~$ rosrun tf tf_echo /map /base_link 
At time 1551951772.615
- Translation: [0.327, -0.494, -0.254]
- Rotation: in Quaternion [0.002, 0.007, -0.518, 0.856]
            in RPY (radian) [-0.004, 0.015, -1.088]
            in RPY (degree) [-0.222, 0.856, -62.336]
At time 1551951772.615
- Translation: [0.327, -0.494, -0.254]
- Rotation: in Quaternion [0.002, 0.007, -0.518, 0.856]
            in RPY (radian) [-0.004, 0.015, -1.088]
            in RPY (degree) [-0.222, 0.856, -62.336]
At time 1551951773.625
- Translation: [0.518, -0.821, -0.254]
- Rotation: in Quaternion [-0.000, 0.004, -0.518, 0.856]
            in RPY (radian) [-0.004, 0.007, -1.088]
            in RPY (degree) [-0.245, 0.399, -62.340]
At time 1551951774.625
- Translation: [0.681, -1.168, -0.254]
- Rotation: in Quaternion [-0.001, 0.005, -0.516, 0.857]
            in RPY (radian) [-0.006, 0.008, -1.083]
            in RPY (degree) [-0.372, 0.467, -62.079]
At time 1551951775.614
- Translation: [0.858, -1.494, -0.254]
- Rotation: in Quaternion [0.000, 0.003, -0.508, 0.861]
            in RPY (radian) [-0.003, 0.006, -1.067]
            in RPY (degree) [-0.152, 0.343, -61.110]
At time 1551951776.624
- Translation: [1.048, -1.834, -0.254]
- Rotation: in Quaternion [-0.002, 0.001, -0.524, 0.852]
            in RPY (radian) [-0.004, -0.001, -1.102]
            in RPY (degree) [-0.249, -0.058, -63.150]
At time 1551951777.619
- Translation: [1.215, -2.160, -0.254]
- Rotation: in Quaternion [0.000, 0.001, -0.514, 0.858]
            in RPY (radian) [-0.001, 0.002, -1.080]
            in RPY (degree) [-0.032, 0.142, -61.884]

```

The orientation of the **`/map`** frame is very critical for me because it's closely related to the `initial_pose` that I'm gonna [set for the `pure_localization` part](https://github.com/googlecartographer/cartographer_ros/issues/579#issuecomment-469160159).  

So I want to know what determines the initial orientation of the **`/map`** frame? 
Any idea for solving this issue?

Thanks a lot !!

MichaelGrupp, 2019-03-14:
Maybe this helps: https://github.com/googlecartographer/cartographer/issues/1507#issuecomment-456813582

NTUwilliam, 2019-03-14:
@MichaelGrupp 
Thanks for replying !

I carefully checked [googlecartographer/cartographer/issues/1507](https://github.com/googlecartographer/cartographer/issues/1507). 
It's a bit relative because I also find that all the global submap poses that lie on the SLAM trajectory in my map do have the **same** `orientation` with the first submap pose, which aligns with the Grid. The `Reference Frame` of the Grid is the `Fixed Frame` (same direction with the `/map` frame), as shown below:
[image]

> Each submap has its own coordinate frame origin that corresponds to the position of the first (accumulated) range scan data that was inserted into it. 
 
By this, do you mean that the orientation of the `/map` frame is determined by the first accumulated range scan data (IMO, equals to the first submap?) ? 
If so, is it correct to say that the orientation of the `/map` frame in the `pure_localization` part is different for every different trial (start position)?

Because my main issue is how to set the **rotation** parameter of the `initial_pose` that I'm gonna [set for the `pure_localization` part](https://github.com/googlecartographer/cartographer_ros/issues/579#issuecomment-469160159), so I'm still not clear about this.

Any hint or advice is much appreciated.

Cheers!

NTUwilliam, 2019-03-17:
-------------------------------------------------------Update-----------------------------------------------------------

I've dug into the `/map` frame and `/base_link` frame a bit more. I found that when I just started the normal **SLAM process** (or **build the .pbstream process**, the same), the `/map(odom)` is overlaid(same) with the `/imu_link`, and the same **x** direction with the `/laser` (as shown below).
[image]

But after around **0.8 second**, the `/base_link` **flipped 180 degrees** and then correspondingly the `/laser` and `/imu_link` also flipped 180 degrees. And the `/base_link` **changed direction** from the `/map(odom)` (as shown below). 
[image]

The whole flipping process is shown in the [flipped.mp4 video](https://entuedu-my.sharepoint.com/:v:/g/personal/zhenyu002_e_ntu_edu_sg/EcddakAiX-1NsjuQqX0AnjgBUAGVdEk1RrFgFDyXuEIKpA?e=gagLMU)

I originally thought it maybe related to the direction of my IMU, so I flipped and changed the direction of the IMU and re-collect the rosbag. But after this the Cartographer even **fails** to build the map. So I don't think it's the issue with my IMU. Then I'm more confused and got two questions:

1. The possible reason why the `/base_link` frame **flip 180 degrees** and changed the direction from the `/map(odom)` frame?
2. Which part of the Cartographer code specify the relation between `/base_link` and `/odom`? Is it possible that the Cartographer code asks the `/base_link` to flip 180 degrees with respect to the `/odom(map)` after receiving the IMU messages? I've browsed through the source code of Cartographer but didn't find the related code. 

Any hint or advice for this?
Thanks!!

@MichaelGrupp @gaschler @neomanic

NTUwilliam, 2019-03-17:
I think I've figured out this issue.
Finally I found the error is in the `backpack_2d.urdf` file. I modified the **joint relation** between `base_link` and `imu_link`, which means I wrongly added the `rpy (roll, pitch, yaw)` relation between them. 

After the correction, `base_link` frame doesn't flip anymore and is the same with the `imu_link` frame. Both of them have the same direction with the `map` frame.

Thanks for all your help!

closed 2019-03-17 by NTUwilliam