cartographer-project/cartographer_ros #1220: Loop closure

fprott, 2019-03-11:
Hello,

I have a rather simple question that I do not seem to find anywhere in the documentation: how to find out that loop-closure reliably happened? 
In other words, if I have a small robot traveling in a circle of a few meters radius, how does cartographer tell me that the circle just has been closed?
Is there a flag or a topic or anything I am missing? If not, what is the easiest way to extract the information?

gaschler, 2019-03-12:
To be sure there is a recent loop closure constraint, you can query `PoseGraphInterface::constraints()`, loop over all constraints and check if there is a recent one of type `INTER` that spans some time.

A simpler solution is to subscribe to the global slam callback `GlobalSlamOptimizationCallback` by setting `PoseGraphInterface::SetGlobalSlamOptimizationCallback`. This will call you back after Cartographer has searched for loop closures. It does not tell you if loop closures were found, it mostly tells you if the background loop closure search is keeping up or falls behind.

fprott, 2019-03-12:
Many thanks. That is very helpful. Also thanks for the quick answer.
Is there a reason to not flag loop closure? Because I find it quite helpful for every robot that is exploring.

closed 2019-03-12 by fprott