cartographer-project/cartographer_ros #1223: Retrieving Landmark IDs

fprott, 2019-03-12:
Hello,

I have another strange question. I use landmarks on my map. Said landmarks have very specific IDs that encodes important information, more specifically what kind of landmark it is.
Now my idea is to retrieve these landmarks and especially the IDs in order to find important way-points on the map. I do this to find none physical points that do not show on the occupancy grid.
So the question: is there a way to retrieve the ID of a landmark from cartographer or are these IDs lost? If so where at which point in the cartographer (so that I can modify the system)?

gaschler, 2019-03-12:
Use `PoseGraphInterface::GetLandmarkPoses`, the IDs are the keys of the returned map.
(Closing as answered.)

closed 2019-03-12 by gaschler

MichaelGrupp, 2019-03-12:
The landmarks are stored with their corresponding unique landmark ID in the pose graph.

There is GetLandmarkPoses() in the pose graph, here's the implementation in 2D: https://github.com/googlecartographer/cartographer/blob/fd67975761e1f7fab60d2a84627b8a8e46d831ff/cartographer/mapping/internal/2d/pose_graph_2d.cc#L916

In cartographer_ros, this also used but the IDs are not exposed in the landmark marker publisher (see [here](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/map_builder_bridge.cc#L362))

MichaelGrupp, 2019-03-12:
oops, @gaschler was faster.

fprott, 2019-03-12:
Many thanks for the quick reply. Please excuse that I didn't found the
issue at first. I am very sorry for this inconvenience.

Mit freundlichen Grüßen

Florian Prott