cartographer-project/cartographer_ros #1224: cartographer bag validation

koelsinha, 2019-03-14:
Hi all,

I am trying to validate my bag with the cartographer validation tool. However I am getting a segmentation fault with it. I know that this issue has been raised in  [#970](https://github.com/googlecartographer/cartographer_ros/issues/970). 
I added a comment there and have not received a response, and hence the new issue. 

I ran the command with gdb orefix and got the following:

```
lead@lead:~/catkin_ws$ rosrun --prefix 'gdb -ex run --args' cartographer_ros cartographer_rosbag_validate -bag_filename testbag_small.bag
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.3) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/lead/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_rosbag_validate...(no debugging symbols found)...done.
Starting program: /home/lead/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_rosbag_validate -bag_filename testbag_small.bag
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
E0314 17:48:13.042814  6817 rosbag_validate_main.cc:286] frame_id "imu_link" on topic /imu has serialization time 1552599846.855815941 but sensor time 1552329229.746757030 differing by -270617 s.
E0314 17:48:13.043143  6817 rosbag_validate_main.cc:286] frame_id "imu_link" on topic /imu has serialization time 1552599846.875946415 but sensor time 1552329229.766777992 differing by -270617 s.
E0314 17:48:13.043187  6817 rosbag_validate_main.cc:286] frame_id "imu_link" on topic /imu has serialization time 1552599846.896587605 but sensor time 1552329229.786935091 differing by -270617 s.

Program received signal SIGSEGV, Segmentation fault.
0x0000000000414014 in cartographer_ros::(anonymous namespace)::Run(std::string const&, bool) ()
(gdb) Quit
(gdb) quit
A debugging session is active.

	Inferior 1 [process 6817] will be killed.

Quit anyway? (y or n) EOF [assumed Y]

```

I am not able to find my core file, although i followed 
[Roslaunch Nodes in Valgrind or GDB](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB)

My ulimit looks like:

```
lead@lead:~/catkin_ws$ ulimit -a
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 30521
max locked memory       (kbytes, -l) 64
max memory size         (kbytes, -m) unlimited
open files                      (-n) 1024
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 0
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 30521
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
```

My sysctl params are:

```
kernel.core_pattern = /home/lead/log/core-%e.%p.%h.%t
kernel.core_pipe_limit = 0
kernel.core_uses_pid = 1

```

You will find the bag I used  [here](https://github.com/koelsinha/Bags/blob/master/testbag_small.bag.zip)

Please let me know if I can provide any more details. 

Thanks in advance!

gaschler, 2019-03-15:
Which version of cartographer are you using? Did you compile the latest GitHub master?

Please compile in mode "CMAKE_BUILD_TYPE=RelWithDebInfo" or "Debug", re-run with GDB, use the command `bt` to print a backtrace and post it here as text.

BTW, there is no need to rosrun, "cartographer_rosbag_validate" is just a regular executable you can run directly with `gdb --args devel/cartographer_ros/lib/.../<exectuble> <arguments>`.

Thank you!

kllysin, 2019-03-19:
@gaschler I tried to install and build the latest cartographer. The build failed and I have summarized it in #1229 . Please let me know if there is a way to fix the build issue.

kllysin, 2019-03-25:
I reinstalled ubuntu and ros and was able to build cartographer and validate my bag.

closed 2019-03-25 by kllysin