cartographer-project/cartographer_ros #1225: cartographer_start_trajectory executable doesn't resolve all possible topics

neomanic, 2019-03-15:
I was trying to get new trajectories started, but eventually figured out that because we are using landmarks, the landmark topic isn't being passed in to the service call options here: https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/start_trajectory_main.cc#L77 , and the new trajectory fails due to an invalid topic. 

We have cartographer running in its own namespace, so the call is 
`rosrun cartographer_ros cartographer_start_trajectory -configuration_directory /home/sorter/ros/our_project/src/our_project/config -configuration_basename 'cartographer_navigate.lua'  -initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0, 0, 0}, rotation = {0., 0., 0., } } }' start_trajectory:=/cartographer/start_trajectory odom:=/odrive/odom scan:=/scan landmark:=/cartographer/landmark`

This works to remap the other topics (since they are resolved in the `Run()`), but fails for the landmark topic.  I suspect the same would be true of the other topics that can be remapped in a cartographer node launch file. (A workaround may be to run cartographer_ros back in the root namespace.)

gaschler, 2019-03-15:
Can't you define the remapping when you start the node in the first place?
Is this a viable work around?

We are planning to re-design and simplify the way how to start a trajectory with a ROS service call.
Please take a look at the design proposal https://github.com/googlecartographer/rfcs/pull/40
In the new proposal, TrajectoryOptions are no longer passed but read from a lua file on the server side. How will this work with your use case?

neomanic, 2019-03-15:
> Please take a look at the design proposal googlecartographer/rfcs#40

I do remap on the initial node start, but it appears this mapping is not carried over to the new trajectory. If I don’t provide the remapping for (eg) the odometry on the service call, it fails as well, even when not running landmarks. 

I’ve been following and support the proposed RFC change to the service call, that would solve the issue too.

MichaelGrupp, 2020-10-17:
Closing because start trajectory is now with the improved interface on master.

closed 2020-10-17 by MichaelGrupp