cartographer-project/cartographer_ros #1228: Compiler problem!

crankler, 2019-03-19:
[12/49] Building CXX object CMakeFiles..._ros.dir/cartographer_ros/ros_map.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /home/hitnuc/cartographer_ws/install_isolated/include -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -I/home/hitnuc/cartographer_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/local/include -isystem abseil/src/abseil -O3 -DNDEBUG    -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o -MF CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o.d -o CMakeFiles/cartographer_ros.dir/cartographer_ros/ros_map.cc.o -c /home/hitnuc/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.cc
In file included from abseil/src/abseil/absl/types/variant.h:77:0,
                 from abseil/src/abseil/absl/hash/internal/hash.h:48,
                 from abseil/src/abseil/absl/hash/hash.h:64,
                 from abseil/src/abseil/absl/container/internal/hash_function_defaults.h:55,
                 from abseil/src/abseil/absl/container/flat_hash_set.h:38,
                 from /home/hitnuc/cartographer_ws/install_isolated/include/cartographer/io/points_batch.h:25,
                 from /home/hitnuc/cartographer_ws/install_isolated/include/cartographer/io/image.h:27,
                 from /home/hitnuc/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.h:24,
                 from /home/hitnuc/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map.cc:17:
abseil/src/abseil/absl/types/internal/variant.h:1091:29: error: ‘absl::type_traits_internal::is_detected_convertible’ has not been declared
 using type_traits_internal::is_detected_convertible;
                             ^
abseil/src/abseil/absl/types/internal/variant.h:1095:23: error: ‘is_detected_convertible’ was not declared in this scope
     absl::conjunction<is_detected_convertible<bool, EqualResult, T>...>::value,
                       ^
abseil/src/abseil/absl/types/internal/variant.h:1095:67: error: template argument 1 is invalid
     absl::conjunction<is_detected_convertible<bool, EqualResult, T>...>::value,
                                                                   ^
abseil/src/abseil/absl/types/internal/variant.h:1095:68: error: expected parameter pack before ‘...’
     absl::conjunction<is_detected_convertible<bool, EqualResult, T>...>::value,
                                                                    ^
abseil/src/abseil/absl/types/internal/variant.h:1095:71: error: template argument 1 is invalid
     absl::conjunction<is_detected_convertible<bool, EqualResult, T>...>::value,
                                                                       ^
abseil/src/abseil/absl/types/internal/variant.h:1100:41: error: ‘is_detected_convertible’ was not declared in this scope
     absl::enable_if_t<absl::conjunction<is_detected_convertible<
                                         ^
abseil/src/abseil/absl/types/internal/variant.h:1101:50: error: template argument 1 is invalid
                           bool, NotEqualResult, T>...>::value,
                                                  ^
abseil/src/abseil/absl/types/internal/variant.h:1101:51: error: expected parameter pack before ‘...’
                           bool, NotEqualResult, T>...>::value,
                                                   ^
abseil/src/abseil/absl/types/internal/variant.h:1101:54: error: template argument 1 is invalid
                           bool, NotEqualResult, T>...>::value,
                                                      ^
abseil/src/abseil/absl/types/internal/variant.h:1106:41: error: ‘is_detected_convertible’ was not declared in this scope
     absl::enable_if_t<absl::conjunction<is_detected_convertible<
                                         ^
abseil/src/abseil/absl/types/internal/variant.h:1107:50: error: template argument 1 is invalid
                           bool, LessThanResult, T>...>::value,
                                                  ^
abseil/src/abseil/absl/types/internal/variant.h:1107:51: error: expected parameter pack before ‘...’
                           bool, LessThanResult, T>...>::value,
                                                   ^
abseil/src/abseil/absl/types/internal/variant.h:1107:54: error: template argument 1 is invalid
                           bool, LessThanResult, T>...>::value,
                                                      ^
abseil/src/abseil/absl/types/internal/variant.h:1112:41: error: ‘is_detected_convertible’ was not declared in this scope
     absl::enable_if_t<absl::conjunction<is_detected_convertible<
                                         ^
abseil/src/abseil/absl/types/internal/variant.h:1113:57: error: template argument 1 is invalid
                           bool, LessThanOrEqualResult, T>...>::value,
                                                         ^
abseil/src/abseil/absl/types/internal/variant.h:1113:58: error: expected parameter pack before ‘...’
                           bool, LessThanOrEqualResult, T>...>::value,
                                                          ^
abseil/src/abseil/absl/types/internal/variant.h:1113:61: error: template argument 1 is invalid
                           bool, LessThanOrEqualResult, T>...>::value,
                                                             ^
abseil/src/abseil/absl/types/internal/variant.h:1118:41: error: ‘is_detected_convertible’ was not declared in this scope
     absl::enable_if_t<absl::conjunction<is_detected_convertible<
                                         ^
abseil/src/abseil/absl/types/internal/variant.h:1119:60: error: template argument 1 is invalid
                           bool, GreaterThanOrEqualResult, T>...>::value,
                                                            ^
abseil/src/abseil/absl/types/internal/variant.h:1119:61: error: expected parameter pack before ‘...’
                           bool, GreaterThanOrEqualResult, T>...>::value,
[... 1170 lines cut ...]
abseil/src/abseil/absl/container/internal/raw_hash_set.h:988:16: error: ‘exchange’ is not a member of ‘absl’
         slots_(absl::exchange(that.slots_, nullptr)),
                ^
abseil/src/abseil/absl/container/internal/raw_hash_set.h:989:15: error: ‘exchange’ is not a member of ‘absl’
         size_(absl::exchange(that.size_, 0)),
               ^
abseil/src/abseil/absl/container/internal/raw_hash_set.h:990:19: error: ‘exchange’ is not a member of ‘absl’
         capacity_(absl::exchange(that.capacity_, 0)),
                   ^
In file included from /home/hitnuc/cartographer_ws/install_isolated/include/cartographer/io/points_batch.h:25:0,
                 from /home/hitnuc/cartographer_ws/install_isolated/include/cartographer/io/points_processor.h:22,
                 from /home/hitnuc/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.h:22,
                 from /home/hitnuc/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/ros_map_writing_points_processor.cc:17:
abseil/src/abseil/absl/container/flat_hash_set.h: At global scope:
abseil/src/abseil/absl/container/flat_hash_set.h:484:8: error: ‘IsUnorderedContainer’ is not a class template
 struct IsUnorderedContainer<absl::flat_hash_set<Key, Hash, KeyEqual, Allocator>>
        ^
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/hitnuc/cartographer_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/hitnuc/cartographer_ws/build_isolated/cartographer_ros && /home/hitnuc/cartographer_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.

crankler, 2019-03-19:
today I try to compile this package as the guide,but there have a few problem.I think maybe it's protoc's problem,su I compile protoc 3.7 for it. At now, I can not make it, so can you give me some advice? Thank you!

nindanaoto, 2019-04-28:
I think the main cause is in Abseil.  So, I recommend you to separately install latest abseil from source. To use separately installed Abseil, change this line(https://github.com/googlecartographer/cartographer_ros/blob/63dce89f6e4222dcb8964697cbd7c22950bbdfd1/cartographer_ros/CMakeLists.txt#L58) to "find_package(absl REQUIRED)".
This idea is from this PR (https://github.com/googlecartographer/cartographer/pull/1570) for cartographer.

MarcelSoler, 2019-05-28:
Hi!
The same happened to me, it didn. I discovered for other ros nodes I had installed ros-kinetic-abseil installed, so if I remove it installation works fine.
I think ros-kinetic-abseil is not the last vesion of abseil and `find_package(Abseil REQUIRED) ` fins the ros one if it's installed.
@pifon2a  I am correct? can this be possible? Can you recommend an fix without uninstalling the pakage?

Thank you

IntelDan, 2019-06-11:
do as what @nindanaoto's instruction, and also remove the ros package abseil, then the problem can be solved.

ojura, 2019-06-11:
Unfortunately this is a known issue: https://github.com/googlecartographer/cartographer_ros/issues/1075

PRs are welcome.

MichaelGrupp, 2020-10-16:
The docs mention this problem now.

closed 2020-10-16 by MichaelGrupp