cartographer-project/cartographer_ros #1229: Failed to build cartographer on ROS

koelsinha, 2019-03-19:
Hi all,

I have been trying to install the latest version of cartographer. I first ran into the error [By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.](https://github.com/koelsinha/Bags/blob/master/test1.txt)

I fixed it by modifying my cmakelist.txt files as follows in both cartographer and cartographer_ros folders:
in CmakeList.txt just replace #find_package(Eigen3 REQUIRED) with the following
find_package(Eigen 3 QUIET)
if(NOT Eigen_FOUND)
include(/usr/share/cmake-2.8/Modules/FindEigen3.cmake)
set(Eigen_INCLUDE_DIR ${EIGEN3_INCLUDE_DIR})
endif()
include_directories(${Eigen_INCLUDE_DIR})

_Originally posted by @MilanMichael in https://github.com/googlecartographer/cartographer/issues/42#issuecomment-352239551_

This took care of the above mentioned issue but now I have a new error [error: ‘SensorId’ is not a member of ‘cartographer::mapping::TrajectoryBuilderInterface’
           ::cartographer::mapping::TrajectoryBuilderInterface::SensorId>& and many more](https://github.com/koelsinha/Bags/blob/master/test2.txt)

I am using ROS indigo on UBUNTU 14.04. 
The cmake version is: 
`lead@lead:~/catkin_ws_cartographer$ cmake --version
cmake version 3.2.2`

The c++ version is 
`/usr/bin/c++ --version
c++ (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`
The result of dpkg -L google-mock is: [gmock.txt](https://github.com/koelsinha/Bags/blob/master/gmock.txt)

Please help.

Thanks in advance!

tyuownu, 2019-03-20:
Maybe you can try the [1.0.0](https://github.com/googlecartographer/cartographer/tree/1.0.0), it more stable and compilable.

kllysin, 2019-03-20:
@tyuownu It guves the exact same errors.

ojura, 2019-03-20:
Cartographer with ROS Indigo on Ubuntu 14.04 is part of the build pipeline which master does pass (and always has). This Docker file builds Cartographer on a clean ROS Indigo 14.04 base: https://github.com/googlecartographer/cartographer_ros/blob/master/Dockerfile.indigo

So something must be wrong with your Ubuntu 14.04 install.

tyuownu, 2019-03-21:
@ojura Cannot agree more, I built it successful on ubuntu 14.04.

kllysin, 2019-03-25:
Thank you everyone for your inputs. I reinstalled ubuntu 14.04 and ros indigo and built the cartographer ros package successfully.

closed 2019-03-25 by kllysin